brocaar/chirpstack-application-server #217: Payload codec and encoder for device

JairoRotava, 2018-02-16:
### Is this a bug or a feature request?
feature
### What did you expect?
Payload decoder/encoder for each device
### What happened?
Payload decoder/encoder is for application. I understand I can use different devices on the same application, so a different decoder/encoder should be used.
### What version are your using?
0.18.0
### How can your issue be reproduced?
Use two different devices under the same project and try to make different decoder/encoder for each one.
### Could you share your log output?

brocaar, 2018-02-18:
I don't think I will implement this. In my opinion, you should have one type of device per application, or use different port numbers for different types of devices / sensors within the same application. Then you are still able to differentiate your decoding per port number using the application encoder / decoder.

Also when having 1000 devices, you don't want to manage the encoding / decoding per device.

fifthaxe, 2018-02-19:
@brocaar: what do you thing about having a concept of sub-application to still be able to group devices in a given parent-application ?

brocaar, 2018-02-19:
Could you elaborate on what the use-case for this would be?

JairoRotava, 2018-02-19:
I missed the port parameter, I can use it. 

I think I confused configuration of Lorawan and Sigfox. On sigfox we are able to decode/encode/callback per device. In that case you can make a list of different decodes and pick one for the device. One suggestion is to include the decode on the device profile, or on something like "decode/encode profile". 

The use case example would be something like this:
- Application: my home automation
   * Devicer 1: Humidity and temp sensor
   * Device 2: Ligths on/off
   * Device 3: Open/Close gate

brocaar, 2018-02-19:
Detaching the encoder / decoder from the application (to make it re-usable) could be an idea. Adding it to the device-profile is probably not the best solution. E.g. OTAA and ABP devices might still share the same encoders / decoders.

closed 2018-02-19 by JairoRotava

fifthaxe, 2018-02-19:
@brocaar beeing able to group devices in a root application

sslupsky, 2018-05-01:
@brocaar Sorry to chime in on a closed issue.  It seems to me that the decoder and encoder should be attached to the device profile?  Each different type of sensor will have a unique data structure for the payload.  Why do you suggest this is not the best solution?