brocaar/chirpstack-application-server #226: Discovery Enable Error

jvillamo, 2018-03-27:
### Is this a bug or a feature request?

### What did you expect?

### What happened?

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?

jvillamo, 2018-03-27:
When I activate the discovery enable, the docker console pulls the following error, every second.

`
postgresql_as_1  | ERROR:  division by zero
postgresql_as_1  | STATEMENT:  
postgresql_as_1  | 			select
postgresql_as_1  | 				g.*
postgresql_as_1  | 			from gateway g
postgresql_as_1  | 			inner join network_server ns
postgresql_as_1  | 				on ns.id = g.network_server_id
postgresql_as_1  | 			where
postgresql_as_1  | 				ns.gateway_discovery_enabled = true
postgresql_as_1  | 				and g.ping = true
postgresql_as_1  | 				and (g.last_ping_sent_at is null or g.last_ping_sent_at <= (now() - (interval '24 hours' / ns.gateway_discovery_interval)))
postgresql_as_1  | 			order by last_ping_sent_at
postgresql_as_1  | 			limit 1
postgresql_as_1  | 			for update
appserver_1      | time="2018-03-27T15:52:37Z" level=error msg="send gateway ping error: get gateway for ping error: select error: pq: division by zero"
`

brocaar, 2018-03-28:
Thanks, that should have been raised a validation error when configuring. For now, please set the interval value at something larger than `0` :-)

commit bd7247c1 (2018-04-04): Validate discovery interval on create / update.

Closes #226.
closed 2018-04-04 by brocaar