brocaar/chirpstack-application-server #228: Automatic Unwanted DataDown Packets

Shiipz, 2018-04-09:
### Is this a bug or a feature request?
Bug or Feature Change as of update.
Recently Upgraded the LoRa App Server, Now finding that every DataUp is generating an automatic DataDown packet addressed to devices from the gateway. Have not changed LoRa Device Code or Setup. All features in LoRa device are off including ADR, we are using UnconfirmedDataUp, ABP joining. 
Anyone know where this setting is or what we are missing to turn off?

### What did you expect?
Expected No DownStream messages from gateway to device pages.

### What happened?
Have UnconfirmedDataDown message every upstream report with DataDown now having ADR true, (ADR still false in DataUp). Also RXParamSetupReq and RXTimingSetupReq in frame option field in DataDown. This has only started once we updated LoRa Server. FOpt is Null in DataUp

### What version are your using?
0.18.2 is our current version,
0.16.x was the previous version.

### How can your issue be reproduced?
Should be clear to see. If not maybe different configs are used. Trying to stop DataDown's.

### Could you share your log output?

brocaar, 2018-04-09:
This is more related to LoRa Server (https://github.com/brocaar/loraserver).

When your devices does not respond to the `RXParamSetupReq` and `RXTimingSetupReq` mac-commands, then this is probably an issue with your device. The network-server tries to update the device configuration to keep it in sync with your `loraserver.toml` file. However it expects an acknowledgement from the device. As it does not receive such acknowledgement, it is resent.

Shiipz, 2018-04-09:
Moved to (https://github.com/brocaar/loraserver).

https://github.com/brocaar/loraserver/issues/318

closed 2018-04-10 by brocaar