brocaar/chirpstack-application-server #234: make api fails with undefined: proto.InternalMessageInfo

urbie-mk2, 2018-05-04:
### Is this a bug or a feature request?
bug / help
### What did you expect?
compiled binary
### What happened?
error message occurs: api/application.pb.go:89:46: undefined: proto.InternalMessageInfo
### What version are your using?
master head
### How can your issue be reproduced?

1. git clone  lora-app-server (this repo)
2. make requirements && make ui-requirements && make api && make
-> error: `api/application.pb.go:89:46: undefined: proto.InternalMessageInfo`

### Could you share your log output?
`api/application.pb.go:89:46: undefined: proto.InternalMessageInfo`

make requirements downloads a complete package in go/src/.. and one incomplete package in lora-app-server/vendor/...

When I delete the protobuf package in vendor lora-app-server compiles. I could not find out from the logs where the vendor protobuf version is coming from but I would like to know and why it is incomplete.

sidenote: lora-app-server compiles if make api is never run before make.

[make-requirements.log](https://github.com/brocaar/lora-app-server/files/1973872/make-requirements.log)

brocaar, 2018-05-04:
Could you also confirm that these steps are failing with the provided Docker development image? See also: https://www.loraserver.io/lora-app-server/community/source/

urbie-mk2, 2018-05-04:
Sorry, lunch break.

It failed as well: 
[lora-app-server-docker-compose.log](https://github.com/brocaar/lora-app-server/files/1974287/lora-app-server-docker-compose.log)

brocaar, 2018-05-07:
This should be fixed. Basically the gRPC generator (newer version) was generating code unsupported by the protobuf library (older version). I've updated the protobuf dependency version.

commit 23d0c403 (2018-05-07): Update protobuf dependecy.

See also: https://github.com/google/protobuf/issues/4582.

Closes #234.
closed 2018-05-07 by brocaar