brocaar/chirpstack-application-server #237: Execution timeout of the downlink program on raspberry

flosers, 2018-05-18:
### Is this a bug or a feature request?
Hello,
I have a bug on a raspberry 3 with lora app server.
### What did you expect?
I made a program for decode the payload receive by the server, the program  work well on my computer but not on lora-app-server.
### What happened?
I have an error "execution timeout" with my  Payload decoder function.
The timeout is maybe too little for the power of a raspberry?
### What version are your using?
I use the 0.20.2 lora-app-server version.
### How can your issue be reproduced?
With a raspberry 

### Could you share your log output?
May 18 08:54:00 raspberrypi lora-app-server[7214]: time="2018-05-18T08:54:00+02:00" level=error msg="decode payload error" application_id=1 codec=CUSTOM_JS dev_eui=XXXXXXXXX error="execution timeout" f_cnt=10 f_port=125

brocaar, 2018-05-18:
Could you share your decoder script?

flosers, 2018-05-18:
Here is the code:
var Base64 = {
_keyStr : "ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz0123456789+/=",
encode : function (input) {
    var output = "";
    var chr1, chr2, chr3, enc1, enc2, enc3, enc4;
    var i = 0;
    input = Base64._utf8_encode(input);
    while (i < input.length) {
        chr1 = input.charCodeAt(i++);
        chr2 = input.charCodeAt(i++);
        chr3 = input.charCodeAt(i++);
        enc1 = chr1 >> 2;
        enc2 = ((chr1 & 3) << 4) | (chr2 >> 4);
        enc3 = ((chr2 & 15) << 2) | (chr3 >> 6);
        enc4 = chr3 & 63;
        if (isNaN(chr2)) {
            enc3 = enc4 = 64;
        } else if (isNaN(chr3)) {
            enc4 = 64;
        }
        output = output +
        this._keyStr.charAt(enc1) + this._keyStr.charAt(enc2) +
        this._keyStr.charAt(enc3) + this._keyStr.charAt(enc4);
    }
    return output;
},
decode : function (input) {
    var output = "";
    var chr1, chr2, chr3;
    var enc1, enc2, enc3, enc4;
    var i = 0;
    input = input.replace(/[^A-Za-z0-9\+\/\=]/g, "");
    while (i < input.length) {
        enc1 = this._keyStr.indexOf(input.charAt(i++));
        enc2 = this._keyStr.indexOf(input.charAt(i++));
        enc3 = this._keyStr.indexOf(input.charAt(i++));
        enc4 = this._keyStr.indexOf(input.charAt(i++));
        chr1 = (enc1 << 2) | (enc2 >> 4);
        chr2 = ((enc2 & 15) << 4) | (enc3 >> 2);
        chr3 = ((enc3 & 3) << 6) | enc4;

        output = output + String.fromCharCode(chr1);

        if (enc3 != 64) {
            output = output + String.fromCharCode(chr2);
        }
        if (enc4 != 64) {
            output = output + String.fromCharCode(chr3);
        }
    }
    output = Base64._utf8_decode(output);
    return output;

},
decodeToHex: function(input){
	var raw = "";
	for(var i = 0; i < input.length; ++i){
		raw+= (String.fromCharCode(input[i]));
	}
	raw=Base64.decode(input)
	var HEX = '';
	  for ( i = 0; i < raw.length; i++ ) {
	   var _hex = raw.charCodeAt(i).toString(16)
	    HEX += (_hex.length==2?_hex:'0'+_hex);
	  }
	  var op= HEX.toUpperCase();
	  return op
},
_utf8_encode : function (string) {
    string = string.replace(/\r\n/g,"\n");
    var utftext = "";
    for (var n = 0; n < string.length; n++) {
        var c = string.charCodeAt(n);
        if (c < 128) {
            utftext += String.fromCharCode(c);
        }
        else if((c > 127) && (c < 2048)) {
            utftext += String.fromCharCode((c >> 6) | 192);
            utftext += String.fromCharCode((c & 63) | 128);
        }
        else {
            utftext += String.fromCharCode((c >> 12) | 224);
            utftext += String.fromCharCode(((c >> 6) & 63) | 128);
            utftext += String.fromCharCode((c & 63) | 128);
        }
    }
    return utftext;
},
_utf8_decode : function (utftext) {
    var string = "";
    var i = 0;
    var c = c1 = c2 = 0;
    while ( i < utftext.length ) {

        c = utftext.charCodeAt(i);

        if (c < 128) {
            string += String.fromCharCode(c);
            i++;
        }
        else if((c > 191) && (c < 224)) {
            c2 = utftext.charCodeAt(i+1);
            string += String.fromCharCode(((c & 31) << 6) | (c2 & 63));
            i += 2;
        }
        else {
            c2 = utftext.charCodeAt(i+1);
            c3 = utftext.charCodeAt(i+2);
            string += String.fromCharCode(((c & 15) << 12) | ((c2 & 63) << 6) | (c3 & 63));
            i += 3;
        }
    }
    return string;
}
}
function hexToDec(payload,firtsB,lastB){
return parseInt(raw.substr(firtsB,lastB), 16)
}
function startwith(payload,trame){
	var x = "/sub/1"; 
	if (payload.substring(0, trame.length) === trame){
		return true;
	}
	else return false;
}
function report_mesure(raw){
	sum_act_en=hexToDec(raw,16,6)
	sum_react_en=hexToDec(raw,22,6)
	nb_samples=hexToDec(raw,28,4)
	act_en=hexToDec(raw,32,4)
	react_en=hexToDec(raw,36,4)
	return {
			"type": "dataToKura",
			"Energie_active_total": sum_act_en,
			"Energie_reactive_total": sum_react_en,
			"nombre_d_echantillon": nb_samples,
			"Energie_active": act_en,
			"Energie_reactive": react_en
			}
	}
function Decode(fPort, bytes) {
	raw=Base64.decodeToHex(bytes)
	if (startwith(raw,"110A00520000410C")) {
		return report_mesure(raw.toString());
    }
  else return {"null":"null"};
}

flosers, 2018-05-18:
You can use this payloads to test my fonction EQoAUgAAQQwAABL///EAVAAN//U=

flosers, 2018-05-18:
I made a mistake the bytes variable is an array of bytes, I don't need to decode base64.However you return the bytes variables in an object it wil be converted in a base64 string in a JSON

closed 2018-05-18 by flosers