brocaar/chirpstack-application-server #239: Database connection fails if password includes '?'

albtrentadue, 2018-06-01:
I think this is an undesired behavior: I modified the password for loraserver_as postgresql user into: '278wbHH?o1'
After this, app-server will not startup properly with message in the log:

giu 01 14:43:47 iotdev2809 lora-app-server[5354]: time="2018-06-01T14:43:47+02:00" level=error msg="ping database error, will retry in 2s: dial tcp: address tcp/278wbHH: unknown port"

It took some time to understand. After changing '?' into a plain character, app-server would start OK.
It is not obvious that '?' is not accepted, should be accepted or docs should inform not to use it.

I am using version 1.0.1

brocaar, 2018-06-04:
Maybe this could be extra documented, but as you can see in the example it splits the URL from the `sslmode` argument:

> postgres://user:password@hostname/database?sslmode=disable

If you think this could be documented better, feel free to create a pull-request :-)

closed 2018-06-04 by brocaar

albtrentadue, 2018-06-05:
Ok, I will think about how to improve the docs and contribute.
Still I believe that this is a weakness in code.
By looking into the line in the log file generated: "address _tcp/278wbHH_: unknown port", it can be seen that the '?' in my password has caused wrong DSN parsing.

brocaar, 2018-06-05:
The string itself is handled by the https://github.com/lib/pq driver. When you think it is a weakness in the code, I recommend creating a ticket there :-)

albtrentadue, 2018-06-05:
Fair enough. Thanks.