brocaar/chirpstack-application-server #245: Request queuing multiple downlink packets close together causes duplicate fCnt numbers

faulteh, 2018-07-04:
### Is this a bug or a feature request?
Bug

### What did you expect?
When using the /api/devices/{devEUI}/queue POST API with requests very close together, the downlink packets that are queued up should each get different fCnt numbers. The 'getting' of the next fCnt number and saving the record needs to be wrapped in a transaction so that no two queued packets get the same fCnt number.

### What happened?
When calling /api/devices/{devEUI}/queue POST from another app to queue up multiple downlink packets at the same time / very close together, some packets are being given the same fCnt.

This causes ERROR webhooks to be sent for the second packet when all the packets get transmitted next time the end device checks in (class A)
`
" 200 2 "-" "Go-http-client/1.1" "-"
backend_1     | 192.168.0.5 - POST /api/mustang/loraserver/ack HTTP/1.0 200 2 - 0.983 ms
backend_1     | 2018-07-04T05:33:48.039Z DEBUG: 	LORASERVER ERROR Received
backend_1     | { applicationID: '1',
backend_1     |   applicationName: 'mustang1',
backend_1     |   deviceName: 'Sim1',
backend_1     |   devEUI: '02000000efbeadde',
backend_1     |   type: 'DEVICE_QUEUE_ITEM_FCNT',
backend_1     |   error: 'invalid frame-counter',
backend_1     |   fCnt: 43 }
backend_1     | 192.168.0.5 - POST /api/mustang/loraserver/error HTTP/1.0 200 2 - 1.247 ms
nginx_1       | 172.16.1.225 - - [04/Jul/2018:05:33:48 +0000] "POST /api/mustang/loraserver/error HTTP/1.1" 200 2 "-" "Go-http-client/1.1" "-"
backend_1     | 2018-07-04T05:33:48.092Z DEBUG: 	LORASERVER ACK Received
backend_1     | { applicationID: '1',
backend_1     |   applicationName: 'mustang1',
backend_1     |   deviceName: 'Sim1',
backend_1     |   devEUI: '02000000efbeadde',
backend_1     |   reference: '',
backend_1     |   acknowledged: false,
backend_1     |   fCnt: 43 }
`

This occurs even with disabled frame count validation turned on.

### What version are your using?
0.21.1 (R-Pi APT Repo)

### How can your issue be reproduced?
This python3 script with aiohttp package can reproduce:
`
#!/usr/bin/env python3
# Reproduces duplicate fCnt when queuing packets for downlink
# This script causes NUM_PACKETS of many async requests to the queue API endpoint
# Requires python3, asyncio and aiohttp module

import aiohttp
import asyncio
import base64

NUM_PACKETS=10
url = 'https://localhost:8080/api'
username = 'admin'
password = 'admin'
deveui = '02000000efbeadde'

def b64encode(payload):
    return base64.b64encode(bytes(payload)).decode('utf-8')

async def queue_packet(payload, headers):
    async with aiohttp.ClientSession(connector=aiohttp.TCPConnector(verify_ssl=False)) as session:
        async with session.post('%s/devices/%s/queue' % (url, deveui),
                                json={'data': b64encode(payload), 
                                      'confirmed': True}, headers=headers) as resp:
            print(await resp.json())

async def main():
    async with aiohttp.ClientSession(connector=aiohttp.TCPConnector(verify_ssl=False)) as session:
        async with session.post('%s/internal/login' % url,
                                json={'username': username, 
                                      'password': password}) as resp:
            resp_json = await resp.json()
            print(resp_json)
            headers = {'authorization': resp_json['jwt'] }
            # Make some arbitrary data for the packet payload
            for x in range(NUM_PACKETS):
                payload = [y for y in range(0, x*NUM_PACKETS, x+1)]
                asyncio.ensure_future(queue_packet(payload, headers))
    # 10sec should be enough to make these requests
    await asyncio.sleep(10)

loop = asyncio.get_event_loop()
loop.run_until_complete(main())
`

The above script will then cause the device queue to look like:
`
{
  "items": [
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": null,
      "fCnt": 46
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": "AAQI",
      "fCnt": 47
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": "AAI=",
      "fCnt": 48
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": "AAMG",
      "fCnt": 48
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": "AAcOFRwjKjE4",
      "fCnt": 49
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": "AAIEBgg=",
      "fCnt": 49
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": "AAgQGCAoMDhA",
      "fCnt": 50
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": "AAUKDxQZHiM=",
      "fCnt": 50
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": "AAYMEhgeJCow",
      "fCnt": 50
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": null,
      "fCnt": 50
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": "AAQIDBAUGBw=",
      "fCnt": 50
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": "AAkSGyQtNj9I",
      "fCnt": 51
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": "AAMGCQwPEg==",
      "fCnt": 51
    },
    {
      "devEUI": "02000000efbeadde",
      "reference": "",
      "confirmed": true,
      "fPort": 0,
      "data": "AAoUHigyPEZQ",
      "fCnt": 51
    }
  ]
}
`

As you can see, the queue has multiple items with the same fCnt.

### Could you share your log output?

API Results included above in lieu of logs.

Lora-app-server logs on R-Pi in /var/log/lora-app-server are empty? (maybe a separate issue in my Pi image generated from the pi-gen repo).

brocaar, 2018-07-04:
Thanks for reporting! And thanks for the clear description of the problem with logs 👍 I'm going to look into this issue.

brocaar, 2018-07-04:
The issue is that (at least for LoRaWAN 1.0) the FCntDown is in the domain of the network-server, while the scheduling is in the domain of the application-server. What happens when you enqueue a downlink item:

* LoRa App Server makes an API call to LoRa Server to retrieve the next FCntDown (it will retrieve the DeviceSession from Redis + takes the max FCntDown from the device_queue table)
* LoRa App Server encrypts the payload (the FCntDown is an argument of the encryption function)
* LoRa App Server enqueues the payload, again using an API call to LoRa Server

This indeed introduces a risk of a race-condition with the error notification as a result. I don't think in the current setup it is easy to make this transaction as the logic is distributed across two services. For this reason the `AFCntDown` / `NFCntDown` separation was introduced in LoRaWAN 1.1. By having two downlink counters (one for the network-server, one for the application-server), the `AFCntDown` in case of application data is in full control of the application-server (LoRa App Server) and can easily be made transnational as LoRa App Server doesn't have to request the next frame-counter from LoRa Server.

One thing that I can think of which would at least give an early error is to add an unique index on `dev_eui` + `f_cnt` in the `device_queue` table in LoRa Server. Then you would get an error directly from the API instead of in the webhook.

brocaar, 2018-07-04:
Thanks also for also providing a script to reproduce the issue 💯 I get many "bug" reports which I can't reproduce because of lacking information or where it is even unclear if it is a bug report or an environmental issue.

I have a fix for this issue which I will release today (as LoRa App Server v1.0.2).

brocaar, 2018-07-04:
This has been release as v1.0.2. I still have to include this into the upcoming v2 and I will close this issue once it has been merged to the master branch.

closed 2018-07-04 by brocaar

faulteh, 2018-07-04:
Thanks for checking this out so quickly, I'd much rather make a Python script to reproduce the problem that trying to learn Go in an afternoon to fix it myself!

darkfader, 2020-01-10:
I still have this problem. Was it fixed?

commit 6d7a7530 (2018-07-04): Lock device row on downlink enqueue.

Closes #245.