brocaar/chirpstack-application-server #246: build failure

linxzhang, 2018-07-06:
$:~/gopath/src/github.com/brocaar/lora-app-server$ make requirements ui-requirements
unexpected directory layout:
        import path: github.com/brocaar/loraserver/api/common
        root: /home/xxxx/gopath/src
        dir: /home/xxxx/gopath/src/github.com/brocaar/lora-app-server/vendor/github.com/brocaar/loraserver/api/common
        expand root: /home/xxxx/gopath/src
        expand dir: /home/xxxx/gopath/src/github.com/brocaar/lora-app-server/vendor/github.com/brocaar/loraserver/api/common
        separator: /
Installing development tools
Gopkg.lock was already in sync with imports and Gopkg.toml
(1/47) Wrote github.com/garyburd/redigo@v1.4.0
(2/47) Wrote github.com/gorilla/websocket@v1.2.0
(3/47) Wrote github.com/brocaar/loraserver@master
(4/47) Wrote github.com/gopherjs/gopherjs@master
(5/47) Wrote github.com/golang/protobuf@v1.1.0
(6/47) Wrote github.com/jacobsa/crypto@master
(7/47) Wrote github.com/inconshreveable/mousetrap@v1.0
(8/47) Wrote github.com/grpc-ecosystem/go-grpc-middleware@master
(9/47) Wrote github.com/gorilla/context@v1.1
(10/47) Wrote github.com/brocaar/lorawan@master
(11/47) Wrote github.com/dgrijalva/jwt-go@v3.1.0
(12/47) Wrote github.com/eclipse/paho.mqtt.golang@master
(13/47) Wrote github.com/fsnotify/fsnotify@v1.4.7
(14/47) Wrote github.com/elazarl/go-bindata-assetfs@master
(15/47) Wrote github.com/gorilla/mux@v1.6.1
(16/47) Wrote github.com/grpc-ecosystem/grpc-gateway@v1.3.1
(17/47) Wrote github.com/hashicorp/hcl@master
(18/47) Wrote gopkg.in/yaml.v2@v2
(19/47) Wrote github.com/jtolds/gls@v4.2.1
(20/47) Wrote github.com/jmoiron/sqlx@master
(21/47) Wrote github.com/lib/pq@master
(22/47) Wrote github.com/magiconair/properties@v1.7.4
(23/47) Wrote github.com/mitchellh/mapstructure@master
(24/47) Wrote github.com/mmcloughlin/geohash@master
(25/47) Wrote github.com/pkg/errors@v0.8.0
(26/47) Wrote github.com/pelletier/go-toml@v1.1.0
(27/47) Wrote github.com/robertkrimen/otto@master
(28/47) Wrote github.com/rubenv/sql-migrate@master
(29/47) Wrote github.com/satori/go.uuid@v1.2.0
(30/47) Wrote github.com/sirupsen/logrus@v1.0.4
(31/47) Wrote github.com/smartystreets/assertions@1.8.1
(32/47) Wrote github.com/smartystreets/goconvey@1.6.3
(33/47) Wrote github.com/spf13/cast@v1.1.0
(34/47) Wrote github.com/spf13/afero@v1.0.2
(35/47) Wrote github.com/spf13/cobra@master
(36/47) Wrote github.com/spf13/pflag@v1.0.0
(37/47) Wrote github.com/spf13/jwalterweatherman@master
(38/47) Wrote github.com/spf13/viper@master
(39/47) Wrote github.com/tmc/grpc-websocket-proxy@master
(40/47) Wrote google.golang.org/genproto@master
(41/47) Wrote golang.org/x/net@master
(42/47) Wrote golang.org/x/crypto@master
(43/47) Wrote gopkg.in/gorp.v1@v1.7.1
(44/47) Wrote gopkg.in/sourcemap.v1@v1.0.5
(45/47) Wrote google.golang.org/grpc@v1.9.1
(46/47) Wrote golang.org/x/sys@master
(47/47) Wrote golang.org/x/text@master
Installing UI requirements
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.1.2 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

up to date in 15.925s
$:~/gopath/src/github.com/brocaar/lora-app-server$ make clean
unexpected directory layout:
        import path: github.com/brocaar/loraserver/api/common
        root: /home/xxxx/gopath/src
        dir: /home/xxxx/gopath/src/github.com/brocaar/lora-app-server/vendor/github.com/brocaar/loraserver/api/common
        expand root: /home/xxxx/gopath/src
        expand dir: /home/xxxx/gopath/src/github.com/brocaar/lora-app-server/vendor/github.com/brocaar/loraserver/api/common
        separator: /
Cleaning up workspace
xxxx@software4test-Vostro-230:~/gopath/src/github.com/brocaar/lora-app-server$ make api
unexpected directory layout:
        import path: github.com/brocaar/loraserver/api/common
        root: /home/xxxx/gopath/src
        dir: /home/xxxx/gopath/src/github.com/brocaar/lora-app-server/vendor/github.com/brocaar/loraserver/api/common
        expand root: /home/xxxx/gopath/src
        expand dir: /home/xxxx/gopath/src/github.com/brocaar/lora-app-server/vendor/github.com/brocaar/loraserver/api/common
        separator: /
Generating API code from .proto files
xxxx@software4test-Vostro-230:~/gopath/src/github.com/brocaar/lora-app-server$ make test
unexpected directory layout:
        import path: github.com/brocaar/loraserver/api/common
        root: /home/xxxx/gopath/src
        dir: /home/xxxx/gopath/src/github.com/brocaar/lora-app-server/vendor/github.com/brocaar/loraserver/api/common
        expand root: /home/xxxx/gopath/src
        expand dir: /home/xxxx/gopath/src/github.com/brocaar/lora-app-server/vendor/github.com/brocaar/loraserver/api/common
        separator: /
Generating combined Swagger JSON
Generating static files
Generating static files
Running tests
can't load package: package .: no Go files in /home/xxxx/gopath/src/github.com/brocaar/lora-app-server
Makefile:18: recipe for target 'test' failed
make: *** [test] Error 1
xxxx@software4test-Vostro-230:~/gopath/src/github.com/brocaar/lora-app-server$

linxzhang, 2018-07-06:
build loraserver & lora-gateway-bridge  success, but when i build lora-app-server (make test), i met below issue:
can't load package: package .: no Go files in /home/xxxx/gopath/src/github.com/brocaar/lora-app-server

brocaar, 2018-07-09:
Please see https://www.loraserver.io/lora-app-server/community/source/ and https://forum.loraserver.io/t/compiling-the-lora-app-server-source-code/643.

closed 2018-07-09 by brocaar