brocaar/chirpstack-application-server #250: FATA[0000] tls cert and tls key must be set for the external api

david1372, 2018-07-11:
Hi.
I have  an error in running  lora-app-server

> eza@ubuntu:~/work1/src/github.com/brocaar/lora-app-server/build$ ./lora-app-server 
> WARN[0000] No configuration file found, using defaults. See: https://www.loraserver.io/lora-app-server/install/config/ 
> INFO[0000] starting LoRa App Server                      docs="https://www.loraserver.io/" version=2.0.0-alpha.2-2-gd9c0628
> INFO[0000] connecting to postgresql                     
> INFO[0000] setup redis connection pool                  
> INFO[0000] handler/mqtt: TLS config is empty            
> INFO[0000] handler/mqtt: connecting to mqtt broker       server="tcp://localhost:1883"
> INFO[0000] applying database migrations                 
> INFO[0000] handler/mqtt: connected to mqtt broker       
> INFO[0000] handler/mqtt: subscribing to tx topic         qos=0 topic=application/+/device/+/tx
> INFO[0000] migrations applied                            count=0
> INFO[0000] starting application-server api               bind="0.0.0.0:8001" ca-cert= tls-cert= tls-key=
> INFO[0000] starting join-server api                      bind="0.0.0.0:8003" ca_cert= tls_cert= tls_key=
> FATA[0000] tls cert and tls key must be set for the external api 
How can fix this?

brocaar, 2018-08-01:
You should not make this change in `root.go`. Please see https://www.loraserver.io/lora-app-server/install/config/.

closed 2018-08-01 by brocaar

DuyTran2626, 2018-10-30:
How can i fixed it. I had the same issues