brocaar/chirpstack-application-server #254: No rxInfo object in MQTT upload packets after upgrade to v2

germanh1982, 2018-08-01:
Hi everyone

Today after migrating my server (running under docker) to the latest version, using version tag 2 of loraserver, lora-app-server and lora-gateway-bridge, I found the rxInfo object is missing in the upload packets. It was working before the upgrade. This is an example upload with the new version:

application/1/node/e8e1e1000100c849/rx {"applicationID":"1","applicationName":"asset-tracker","deviceName":"SPOTT_AC88","devEUI":"e8e1e1000100c849","txInfo":{"frequency":917000000,"dr":2},"adr":true,"fCnt":284962,"fPort":3,"data":"ACH+AAAAAAAAAAA=","object":{"batt":"100.00","errorcode":"noError","gpsfix":0,"lat":null,"lon":null,"temp":33}}

As a sidenote, I let the server do the database migration by itself (automigrate=true). Also migrated to the new configuration file, just changing the jwt_secret variable, to check if it was a configuration issue.

Do you know what else can I check in order to solve this issue?
Thanks

simonnilsson, 2018-08-02:
I believe you can set this in your service-profile.

[image]

schrein, 2018-08-09:
Same problem after migration to v2, rxInfo object disappear in application rx. As mentioned by @simonnilsson  I just had to re apply option _Add gateway meta-data_ in service profile despite the fact it was already checked.

brocaar, 2018-08-11:
Thanks again. I've been able to track down the issue. It has to do with a cache change in LoRa Server v1 -vs- v2. Saving the form again invalidates the cache and solves the issue (as you have noticed). I'll fix this and issue a bugfix release soon (expect somewhere next week).

closed 2018-08-13 by brocaar