brocaar/chirpstack-application-server #258: Packages for CentOS/RHEL-based distro's

proffalken, 2018-08-20:
### Is this a bug or a feature request?

Feature Request

### What did you expect?

To install via YUM

### What happened?

I wrote my own Ansible playbooks to install the binaries from the download URL

### What version are your using?

2.0.2

### How can your issue be reproduced?

Try to install on CentOS

### Could you share your log output?

N/A

### More details

I've noticed that there are .deb packages, but not RPM's.  Is this something that the community would be interested in having, and if so, would loraserver.io be able to host them in the same way that they currently host the .deb packages?

*NOTE:* I'm willing to do the work to get this done across LoRaServer and LoRa-App-Server, I just want to confirm that it's a useful thing to add to the project before I do so!

brocaar, 2018-08-20:
Yes, this would absolutely be welcome and I would be more than happy to host this :-)

 Please note that eventually I will migrate the `fpm` based `.deb` packaging over to `goreleaser` (which is already being used to generate the `.tar.gz` files). `goreleaser` provides a tool called `nfpm` which can be used to generate `.rpm` and `.deb` packages. That way no other tooling is needed to generate these packages.

(`goreleaser` is a tool to build and distribute Go projects, see also: https://github.com/brocaar/lora-app-server/blob/master/.goreleaser.yml for the lora-app-server configuration)

If we could come up with something generic scripts that could be used for both Debian and Redhat / CentOS based systems (supporting init and systemd), that would be great as this would make it easier to maintain. These scripts can be configured in goreleaser using:

```yaml
  scripts:
    preinstall: "scripts/preinstall.sh"
    postinstall: "scripts/postinstall.sh"
    preremove: "scripts/preremove.sh"
    postremove: "scripts/postremove.sh"
```

See also https://goreleaser.com/customization/#NFPM

proffalken, 2018-08-20:
Ah, ok, thanks, that looks awesome.  I'm familiar with FPM, but I'm still getting to grips with GOLANG.
I'll checkout goreleaser and see if I can come up with something.

brocaar, 2018-09-12:
Please see: https://github.com/brocaar/lora-geo-server/blob/master/.goreleaser.yml and https://goreleaser.com/customization/#NFPM.

By changing

```yaml
formats:
  - deb
```

To

```yaml
formats:
  - deb
  - rpm
```

This might already work (e.g. run `make snapshot`) to generate rpms. What would then be left is to test if the configured scripts and files in the above link will also work on CentOS / RedHat / make changes where needed.

Eventually, my plan is to start using this goreleaser feature for loraserver, lora-app-server and lora-gateway-bridge as it simplifies the release flow :)

proffalken, 2018-09-12:
Oh, cool, thanks. I've not had time to get to this yet, I'm hoping to do so in the next couple of days.

brocaar, 2018-10-10:
In the meanwhile I've cleaned up the (.deb) packaging of all components to use `goreleaser`. If you have some time to test the `rpm` package format (see my previous comment), that would be great!

dtony, 2019-02-15:
Hi, I just successfully build and installed `.rpm` for loraserver and lora-app-server. They build, install and startup fine (systemd) on Centos 7. I didn't tested everything intensively yet. My packages are name like : `lora-app-server_SNAPSHOT-none_linux_amd64.rpm` and when installing it is listed as version 0.0.0 . I am new to `goreleaser`, proper options could probably help.

If you could add the rpm builds for the next release or even better add rpm for the latest release it would be awesome !

dtony, 2019-09-19:
@brocaar have you made any progress on using goreleaser and addind rpm ? Is there anything missing on which we could help ?

Best regards.

brocaar, 2019-11-13:
If one could do some testing with the goreleaser generated `.rpm` packages, that would be great :-)

See also:
* https://goreleaser.com/nfpm/
* https://github.com/brocaar/chirpstack-application-server/blob/master/.goreleaser.yml

dtony, 2019-11-14:
I made the pull requests, you can find them here :
https://github.com/brocaar/chirpstack-network-server/pull/454
https://github.com/brocaar/chirpstack-application-server/pull/383
https://github.com/brocaar/chirpstack-gateway-bridge/pull/145
https://github.com/brocaar/chirpstack-geolocation-server/pull/3

proffalken, 2019-11-15:
Thanks folks, and apologies I've not been able to focus more on this :(

brocaar, 2020-01-19:
This has been done :)

closed 2020-01-19 by brocaar