brocaar/chirpstack-application-server #261: Class-C Downlinks being sent to 'gateway/0000000000000000/tx' with no tx details

sa-wilson, 2018-08-28:
### What happened?
I'm attempting to send a downlink message to a class-c device (that has joined and transmitted once quite recently) via MQTT.
The app server is processing the JSON and queuing a down-link, which is then being sent immediately, but with the following JSON structure, to the `gateway/0000000000000000/tx` MQTT endpoint.
```
{"token":47484,"txInfo:{"mac":"0000000000000000","immediately":false,"frequency":0,"power":0,"dataRate":{"modulation":""},"codeRate":"","iPol":null,"board":0,"antenna":0},"phyPayload":"YA8iAPyKBgADAAgAcAMh/wARZOmWdRtygfM="}
```
This results in no downlink being sent to the device.
### What version are your using?
lora-app-server 2.0.1
loraserver 2.0.2
### LoRa App Server Log Output
```
Aug 28 10:19:08 loraserver.local lora-app-server[25430]: time="2018-08-28T10:19:08+10:00" level=info msg="handler/mqtt: data-down payload received" topic=application/26/device/deadc0de006500ff/tx
Aug 28 10:19:08 loraserver.local lora-app-server[25430]: time="2018-08-28T10:19:08+10:00" level=info msg="finished client unary call" grpc.code=OK grpc.method=GetNextDownlinkFCntForDevEUI grpc.service=ns.NetworkServerService grpc.time_ms=2.238 span.kind=client system=grpc
Aug 28 10:19:08 loraserver.local lora-app-server[25430]: time="2018-08-28T10:19:08+10:00" level=info msg="finished client unary call" grpc.code=OK grpc.method=CreateDeviceQueueItem grpc.service=ns.NetworkServerService grpc.time_ms=5.203 span.kind=client system=grpc
Aug 28 10:19:08 loraserver.local lora-app-server[25430]: time="2018-08-28T10:19:08+10:00" level=info msg="downlink device-queue item handled" confirmed=false dev_eui=deadc0de006500ff f_cnt=8
```
### LoRaServer Log Output
```
Aug 28 10:19:08 loraserver.local loraserver[3085]: time="2018-08-28T10:19:08+10:00" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=GetNextDownlinkFCntForDevEUI grpc.service=ns.NetworkServerService grpc.start_time="2018-08-28T10:19:08+10:00" grpc.time_ms=1.433 peer.address="[::1]:43238" span.kind=server system=grpc
Aug 28 10:19:08 loraserver.local loraserver[3085]: time="2018-08-28T10:19:08+10:00" level=info msg="device-queue item created" dev_eui=deadc0de006500ff f_cnt=8
Aug 28 10:19:08 loraserver.local loraserver[3085]: time="2018-08-28T10:19:08+10:00" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=CreateDeviceQueueItem grpc.service=ns.NetworkServerService grpc.start_time="2018-08-28T10:19:08+10:00" grpc.time_ms=4.783 peer.address="[::1]:43238" span.kind=server system=grpc
Aug 28 10:19:08 loraserver.local loraserver[3085]: time="2018-08-28T10:19:08+10:00" level=info msg="device-queue deleted" id=47
Aug 28 10:19:08 loraserver.local loraserver[3085]: time="2018-08-28T10:19:08+10:00" level=info msg="pending mac-command block set" cid=LinkADRReq commands=2 dev_eui=deadc0de006500ff
Aug 28 10:19:08 loraserver.local loraserver[3085]: time="2018-08-28T10:19:08+10:00" level=info msg="backend/gateway: publishing tx packet" qos=0 topic=gateway/0000000000000000/tx
Aug 28 10:19:08 loraserver.local loraserver[3085]: time="2018-08-28T10:19:08+10:00" level=info msg="device-session saved" dev_addr=fc00220f dev_eui=deadc0de006500ff
```

### JSON Sent to LoRa App Server
```
{
    "fPort": 100,
    "reference": "testtt",
    "confirmed": false,
    "object": {
        "str": "adc"
    }
}
```

### Payload Encoder
```
function Encode(fPort, obj) {
  var bytes = [];
  if (fPort == 100) {
    for (var i = 0; i < obj.str.length; ++i) {
      bytes.push(obj.str.charCodeAt(i));
    }
  }
  return bytes;
}
```

sa-wilson, 2018-08-28:
I've been looking into the code and it seems like this may be happening in LoRaServer, during `(DeviceSession) GetDownlinkGatewayMAC` called by `getDataTXInfoForRX2`, in turn called by `HandleScheduleNextQueueItem`.

It seems like the gateway is not getting found, and it may be returning on line `322` of `internal/downlink/data/data.go`, but not short circuiting the loop in `HandleScheduleNextQueueItem`, and not reporting an error in the logs.

I wasn't sure when I posted the issue wether this was a problem in the app server or network server, but it definitely looks like the network server.

brocaar, 2018-08-28:
Would you mind re-creating this issue at the loraserver repository? This seems indeed more related to LoRa Server than LoRa App Server.

sa-wilson, 2018-08-28:
Done. I thought it may have been an app-server issue at first until I looked into it more.

brocaar/loraserver#351

closed 2018-08-28 by sa-wilson