brocaar/chirpstack-application-server #263: live lorawan frames get,live device data not

j847793745, 2018-09-14:
### Is this a bug or a feature request?
feature request
### What did you expect?
how to find out what happened
### What happened?
gateway live lorawana frames display device uplink data, but device live device data not. What might be the reason, how to locate?
### What version are your using?
Docker install the latest
### How can your issue be reproduced?
n/a
### Could you share your log output?
[image]
[image]

brocaar, 2018-09-17:
This is not a bug or feature request. Please see https://forum.loraserver.io. This question has been asked on the forum before and you'll find hints which might help you to solve this issue.

closed 2018-09-17 by brocaar

j847793745, 2018-09-21:
@brocaar thanks. In fact, I did find a solution, checked disable the frame counter validation, but it was only abp,  and otaa still not work, can't be activated. 
And loraserver log normal, lora-app-server log shows the following: 
level=error msg="processing rx packet error: 
get device-session error: device-session does not exist or invalid fcnt or mic" 

about otaa, how can I solve this?