brocaar/chirpstack-application-server #266: MQTT notification right after JoinAccept

tomasol, 2018-09-28:
Feature Request:
The /join MQTT notification is sent only when device sends first uplink. What I need is a notification right after JoinAccept is sent. Is it possible to add this as a new notification? If I try to implement it, would you accept the change? I envision something like 
Topic: application/[applicationID]/device/[devEUI]/joinaccept
Content: same as /join

brocaar, 2018-09-28:
This is not possible because there are two types of joins: join and rejoin. To support both joins in a transparent way, the new security context is always sent to the application-server on the first uplink (which is when you see the `/join`).

This is the first moment where the network-server can validate that the device actually processed the join-accept message. In case of a rejoin-request, the network-server and application-server must use the old security context until it has received the first uplink using the new security-context (LoRa Server will keep both). This is to make sure that on a rejoin where the join-accept message was not received by the device, the device can continue to operate using the old security context.

Please refer to the LoRaWAN specification (LoRaWAN 1.1) for more information about the rejoin-request.

tomasol, 2018-10-01:
Understood. However for my use case - packet filtering on gateway, I need to unblock the newly assigned address before the device can send anything. Semantic of `/join` is that device actually used the assigned keys/devAddr, so `/joinaccept` or other named event (joinproposed?) would mean AS proposed this devAddr. Consumer of this notification would have to accept the fact that the device might still be using the old security context.

brocaar, 2018-10-01:
Could you share the use-case behind packet filtering on the gateway?

What could be possible is to implement your own join-server interface (or proxy between LoRa Server and the LoRa App Server join-server API). Then you can, as soon as the join-request comes in and is authenticated by the join-server provision the issued DevAddr into the gateway.

Currently I'm not sure if it makes sense to change the LoRa App Server `/join` notification for this as it adds some extra complexity. Also, the application-server is more in the domain of the end-application, the network-server (LoRa Server) is more in the domain of the network. Therefore I recommend to keep these kind of management related things as close to the network-server as possible instead of the application server.

tomasol, 2018-10-02:
I need to maintain a list of DevAddresses that are allowed to communicate. Ordered by effort, I believe I have those 3 options: fork AS and add the notification, listen on bridge MQTT messages and decipher JoinAccept, implement join server.

brocaar, 2018-10-02:
Why not leave this to the network-server? When the device-activation is removed, it will automatically be rejected by the network-server.

brocaar, 2018-10-03:
I think this use-case is too specific to solve in the LoRa Server code-base. I think it would be best to implement a really simple proxy between the LoRa Server and LoRa App Server join-interface. The `JoinReq` and `RejoinReq` will expose all the info needed (DevEUI and DevAddr).

An other solution would be that your keep the `/join` integration, but that your gateway always forwards unknown `DevAddr` payloads with `FCnt=0`.

closed 2018-10-03 by brocaar

tomasol, 2018-10-03:
Thanks, I really like the join server proxy. The fcnt=0 is also clever, but might be fragile if the packet loss is high. This issue can be closed.