brocaar/chirpstack-application-server #267: battery level display not a percentage

daltonate, 2018-10-02:
### Is this a bug or a feature request?
I believe this is a bug.

### What did you expect?
The battery level shown in percentage

### What happened?
The battery level is displayed as provided by the device (i.e. between 1 and 254), this should be scaled in order to show an actual percentage, or the percent sign might be removed for consistency.
In case the percentage semantic is kept, I'm not sure wether the scaling should be done in the network server or in the app server.

### What version are your using?
2.2.0

### How can your issue be reproduced?
Activate the 'Report device battery level to application-server' and look at the reported battery level shown in the application page ( with the devices list)

### Could you share your log output?
Don't think it is useful.

Thanks !

brocaar, 2018-10-03:
Yes, you might be right. The exposed values make sense in the context of the network-server, but I think it is indeed better to expose these as a percentage to the application-server as the 1..254 value requires knowledge about the LoRaWAN protocol.

daltonate, 2018-10-03:
Thanks for the fix !

commit fda95bae (2018-10-08): Update battery-status to hold percentage.

Closes #267.
closed 2018-10-08 by brocaar