brocaar/chirpstack-application-server #276: Adding activation keys to device problem

golovlevviktor, 2018-11-25:
### Is this a bug or a feature request?
bug 

### What did you expect?
I wanted to add new device via http API (_As feature it will be cool if it will be possible to add device and set activation keys with only one request:)_ )

### What happened?
In `lora-app-server-host/api` section there is a sample of the data to `/api/devices/{device_keys.dev_eui}/keys` request:
> {
  "deviceKeys": {
    "appKey": "string",
    "devEUI": "string",
    "nwkKey": "string"
  }
}

I set OTAA activation in `device-profile` and for some reason `appKey` request was successful only when i set `nwkKey` and that what I wrote in **nwkKey** become a **appKey** in web interface!
Have somebody have the same thing? Or Im doing smth wrong? 
Thanks!

### What version are your using?

2.4.0

### How can your issue be reproduced?

Yes, All can be done on `/api` page

### Could you share your log output?

There is no usefull, as I saw

brocaar, 2018-12-11:
This is not a bug but a presentation "feature". Note that in the LoRaWAN 1.1 specification, the AppKey has been renamed to NwkKey. This change has been applied to the API. However, when you select LoRaWAN 1.0.x in the device-profile, the the web-interface will make this translation from App Key to Nwk Key for you.

See also:

https://github.com/brocaar/lora-app-server/blob/master/api/device.proto#L200
https://github.com/brocaar/lora-app-server/blob/master/ui/src/views/devices/DeviceKeys.js#L53

closed 2018-12-11 by brocaar

golovlevviktor, 2018-12-13:
Thanks for your explanation!