brocaar/chirpstack-application-server #277: log rotation issue

gillespilloudwyres, 2018-12-03:
On my loraserver instance, when my logs are rotated, based on the pre-defined logrotate process. a new zip file is created at each rotation, but the original file is conserved and the disk space is not saved.

I think this stackoverflow post is related : https://serverfault.com/questions/221337/logrotate-successful-original-file-goes-back-to-original-size

Should I use a -HUP value to help ?

`gpilloud@loraserver2:/etc/logrotate.d$` sudo logrotate -f -v lora-app-server
`reading config file lora-app-server
`
`Handling 1 logs
`
`rotating pattern: /var/log/lora-app-server/lora-app-server.log  forced from command line (7 rotations)
`empty log files are rotated, old logs are removed
`considering log /var/log/lora-app-server/lora-app-server.log
`  log needs rotating
`rotating log /var/log/lora-app-server/lora-app-server.log, log->rotateCount is 7
`dateext suffix '-20181203'
`glob pattern '-[0-9][0-9][0-9][0-9][0-9][0-9][0-9][0-9]'
`copying /var/log/lora-app-server/lora-app-server.log to /var/log/lora-app-server/lora-app-server.log-`20181203
`truncating /var/log/lora-app-server/lora-app-server.log
compressing log with: /bin/gzip

During the rotation the lora-app-server.log is reduced to 0 but at the end, this size back to his original size.

This problem seems impacting also the others loraserver componants.

brocaar, 2018-12-09:
I can confirm this is an issue for LoRa Server installations **that do not use systemd**.

I could use your help here. I was in the understanding that `copytruncate` would do the trick, but as you found out this is not the case. `/etc/init.d/lora-app-server` contains the following snippet to start the process:

```bash
function do_start {
	start-stop-daemon --start --background --chuid "$DAEMON_USER:$DAEMON_GROUP" --make-pidfile --pidfile "$PID_FILE" --startas /bin/bash -- -c "exec $DAEMON > /var/log/$NAME/$NAME.log 2>&1"
}
```

(I believe the `>` should also be `>>` to append to an existing file)

What would be the right way to logrotate, without the need to restart the `lora-app-server` process?

gillespilloudwyres, 2018-12-09:
After reading this article https://superuser.com/questions/291368/log-rotation-of-stdout it seems to be a good idea.

I will check that this week.

gillespilloudwyres, 2018-12-10:
Operations done today : 
1. init.d script upgrade
2. lora-app-server restart
3. logrotate forced by `sudo logrotate -v -f lora-app-server`

The log has been truncated by the operation

Now, will wait the tomorrow automatic log rotate to close this validation. It seems ok. If it's ok, I will purpose a merge request.

sslupsky, 2018-12-14:
I am seeing a similar log rotate issue on macOS.  When the log file is turned over, the new log file no longer receives any of the log data.  All applications (server, app server and gateway) are affected.

gillespilloudwyres, 2018-12-14:
A PR has been purpose on the init.d scripts (debian) the cause is the stream still opened to the log file when using the > redirect. The >> has fixed my issue.

sslupsky, 2018-12-14:
I am using macOS.  I have a launch daemon running the applications.  The launchd file has the following:

  <key>StandardOutPath</key>
  <string>/usr/local/var/log/lora-app-server.log</string>
  <key>StandardErrorPath</key>
  <string>/usr/local/var/log/lora-app-server.log</string>

I found an article here (https://stackoverflow.com/questions/13207365/how-can-i-add-arguments-to-osx-launchd-plist-to-pipe-output-to-a-log-file) that suggests the files are appended.

brocaar, 2018-12-18:
Thanks @gillespilloudwyres I've merged your PRs :-)

closed 2018-12-18 by brocaar