brocaar/chirpstack-application-server #278: jwt authentication error in live data page

mikumaycry, 2018-12-05:
### Is this a bug or a feature request?
bug
### What did you expect?
To make live data tools works normally
### What happened?
error infos keeps flashing on screen when using live data tool and websocket is not connected
[image]
### What version are your using?
lora-app-server v2.4.0 with chrome 70.0.3538.110
### How can your issue be reproduced?
This issue is not promised to be reproduced, just use the live data page and error keeps flashing, while other apis works normally.
### Could you share your log output?
```
time="2018-12-05T02:14:16Z" level=info msg="finished streaming call with code Unauthenticated" error="rpc error: code = Unauthenticated desc = authentication failed: jwt parse error: json: cannot unmarshal string into Go struct field Claims.exp of type int64" grpc.code=Unauthenticated grpc.method=StreamEventLogs grpc.service=api.DeviceService grpc.start_time="2018-12-05T02:14:16Z" grpc.time_ms=0.14 peer.address="127.0.0.1:45638" span.kind=server system=grpc
time="2018-12-05T02:14:16Z" level=info msg="finished streaming call with code Unauthenticated" error="rpc error: code = Unauthenticated desc = authentication failed: jwt parse error: json: cannot unmarshal string into Go struct field Claims.exp of type int64" grpc.code=Unauthenticated grpc.method=StreamEventLogs grpc.service=api.DeviceService grpc.start_time="2018-12-05T02:14:16Z" grpc.time_ms=0.139 peer.address="127.0.0.1:45638" span.kind=server system=grpc
```

brocaar, 2018-12-09:
are you using apache, nginx or any other proxy?

brocaar, 2019-07-02:
Can this issue be closed?

closed 2019-07-04 by mikumaycry