brocaar/chirpstack-application-server #283: Feature request:  Add gateway MAC address as tag to influx integration

sslupsky, 2018-12-14:
Feature request

I would like to be able to track which gateway the device data was received from when using the InfluxDB integration.  Would it be possible to have the option to tag the device data with the gateway MAC address?

brocaar, 2018-12-18:
This would be an array of tags as the data could be received by multiple gateways. I don't think InfluxDB has support for storing an array value for tags.

sslupsky, 2018-12-18:
Can you add multiple tags?

Or, the server selects which gateway to "accept" the data from?  So, perhaps we can just tag that one?

sslupsky, 2018-12-18:
Also, I have another situation where I would like to add a tag to the data going to influx.  I could not figure out a way to do that using a custom codec.  Specifically, I would like to add a geohash (or lat / lon) based on the device eui or device name (which themselves are added as other tags).  It would be helpful if there was a capability to add this type of conditional metadata to the data on the way to the database.  Something like the custom codec, but specifically for meta data (tags).

brocaar, 2019-09-16:
I'm closing this issue. InfluxDB doesn't support an array of tag values, but you could use the PostgreSQL integration instead. Meta-data tags have been added a while ago. You will find these under the device configuration.

closed 2019-09-16 by brocaar