brocaar/chirpstack-application-server #290: error in react and ui

wssq, 2019-01-28:
Hi.
When i want to use npm for debugging lora-app-server, I get fallow error. Of course It is not problem in using build.
←→1 of 2 errors on the page
Unhandled Rejection (TypeError): Cannot read property 'code' of undefined
errorHandler
src/github.com/brocaar/lora-app-server/ui/src/stores/helpers.js:23
  20 |     },
  21 |   });
  22 | } else {
> 23 |    if (error.response.obj.code === 16) {
     | ^  24 |     history.push("/login");
  25 |      } else {
  26 |       dispatcher.dispatch({
View compiled
This screen is visible only in development. It will not appear if the app crashes in production.
Open your browser’s developer console to further inspect this error.

wssq, 2019-02-05:
Can anyone help me?

brocaar, 2019-02-28:
I don't get this error. What should I do to reproduce your issue?

closed 2019-03-20 by brocaar

gloffreda, 2021-02-18:
Got same error here.

- cloned repo
- started http-server on static folder (now i can browse localhost:8080 and get swagger
- npm start on ui folder

[image]

Getting these 2 not found on page loading and the above error on main screen

[image]

Thanks for helping