brocaar/chirpstack-application-server #292: Application session key invalid on first frame after join

guillaumebignon, 2019-01-31:
### Is this a bug or a feature request?
Bug

### What did you expect?
I'm trying to send a frm payload to my device on the first frame after join

### What happened?
Data received by my device isn't the good one (length is ok but bytes are not).
After a few tests, I noticed all frames added to the queue between the join accept and the first uplink are invalid when received by the device.
The Application session key field on activation view is updated on the first uplink (after the join request).

### What version are your using?
2.4.1

### How can your issue be reproduced?
Every time I publish a frame via mqtt before the first uplink, the received frame is wrong.

jamesl-dm, 2019-02-01:
I'm seeing this too, on LoRaWAN 1.1

brocaar, 2019-05-06:
This has been fixed by the last LoRa App Server / LoRa Server release. It will detect when the security-context is out-of-sync and in case so, will reject the enqueue.

closed 2019-05-06 by brocaar