brocaar/chirpstack-application-server #293: gRPC failed in selfbuild version

bobbyac01, 2019-02-01:
### Is this a bug or a feature request?
Bug
### What did you expect?
[image]
Show the gateway information normally
### What happened?
It starts with compiling problem ProtoPackageIsVersion3
I've read and follow these two discussion to make it compile but the gRPC seems broke due to packages dependency problem.
https://forum.loraserver.io/t/complling-lora-app-server-source-with-docker-gives-proto-protopackageisversion3-error/3147
https://github.com/golang/protobuf/issues/763
### What version are your using?
Fresh installed Ubuntu 18.04 and docker-compose 1.23.2, docker 18.09.1.
### How can your issue be reproduced?
Install Ubuntu 18.04, docker-compose 1.23.2, docker 18.09.1 and git clone this repository
Then follow https://www.loraserver.io/lora-app-server/community/source/
First you will face ProtoPackageIsVersion3 when "make build"
Then you "git checkout v1.2.0" the golang/protobuf/protoc-gen-go & "go install"
"make clean", "make api", "make build" it will finished the compile.
But when you using the web ui will found out it's gRPC failed in some items.
### Could you share your log output?
time="2019-01-24T12:21:43Z" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=List grpc.service=api.GatewayService grpc.start_time="2019-01-24T12:21:43Z" grpc.time_ms=266.357 peer.address="127.0.0.1:36771" span.kind=server system=grpc
time="2019-01-24T12:21:43Z" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=List grpc.service=api.OrganizationService grpc.start_time="2019-01-24T12:21:43Z" grpc.time_ms=505.406 peer.address="127.0.0.1:36773" span.kind=server system=grpc
time="2019-01-24T12:21:43Z" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=List grpc.service=api.OrganizationService grpc.start_time="2019-01-24T12:21:43Z" grpc.time_ms=263.054 peer.address="127.0.0.1:36773" span.kind=server system=grpc
time="2019-01-24T12:21:44Z" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=Get grpc.service=api.OrganizationService grpc.start_time="2019-01-24T12:21:43Z" grpc.time_ms=867.933 peer.address="127.0.0.1:36773" span.kind=server system=grpc
time="2019-01-24T12:21:44Z" level=warning msg="creating insecure network-server client" server="localhost:8000"
time="2019-01-24T12:21:44Z" level=info msg="finished client unary call" error="rpc error: code = InvalidArgument desc = timestamp: nil Timestamp" grpc.code=InvalidArgument grpc.method=GetGatewayStats grpc.service=ns.NetworkServerService grpc.time_ms=42.302 span.kind=client system=grpc
time="2019-01-24T12:21:44Z" level=info msg="finished unary call with code InvalidArgument" error="rpc error: code = InvalidArgument desc = timestamp: nil Timestamp" grpc.code=InvalidArgument grpc.method=GetStats grpc.service=api.GatewayService grpc.start_time="2019-01-24T12:21:43Z" grpc.time_ms=1066.532 peer.address="127.0.0.1:36771" span.kind=server system=grpc
time="2019-01-24T12:21:44Z" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=Get grpc.service=api.OrganizationService grpc.start_time="2019-01-24T12:21:44Z" grpc.time_ms=286.21 peer.address="127.0.0.1:36773" span.kind=server system=grpc
time="2019-01-24T12:21:44Z" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=List grpc.service=api.OrganizationService grpc.start_time="2019-01-24T12:21:44Z" grpc.time_ms=317.839 peer.address="127.0.0.1:36773" span.kind=server system=grpc
time="2019-01-24T12:21:44Z" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=Get grpc.service=api.OrganizationService grpc.start_time="2019-01-24T12:21:44Z" grpc.time_ms=58.086 peer.address="127.0.0.1:36773" span.kind=server system=grpc
""rpc error: code = InvalidArgument desc = timestamp: nil Timestamp"" happened when clicking into gateway information

### My research
It seems start with protobuf's update. Then it cause lora-app-server packages dependency problem.
I find some discussion but no solution for it.
Thanks for your viewing and your helping hand.

id-db, 2019-02-04:
If it helps, try not to build the proto files (leave out make api) for a temporary workaround.

brocaar, 2019-02-13:
I do not consider this a bug. I expect that this is due to a protobuf library and protoc-gen-go compatibility issue since the latter one has been updated since I have generated the API code for the last time. As @id-db already mentioned, the `make api` is not a required step and only needs to be executed when you are making API changes.

When you would like to work on bringing protoc-gen-go and protobuf libary in sync, I would welcome a pull-request.

> ""rpc error: code = InvalidArgument desc = timestamp: nil Timestamp"" happened when clicking into gateway information

If this issue would be with the pre-compiled binaries, then I would consider this a bug. However I can not support self compiled binaries.

bobbyac01, 2019-02-20:
Ok I got it work after ignoring make api.
Thanks for yours reply.
Seems we can't change the API by ourselves if still using protobuf library before someone give the right configuration.

brocaar, 2019-02-20:
@bobbyac01 actually, I have made the first step towards more consistent protobuf generation in https://github.com/brocaar/lora-app-server/commit/581519f8947add528b508895f5c8b4ce6b89f7f2. By switching to Go modules instead of Dep, I can also pinpoint tools like the Go protobuf generator to a specific version. Probably a bit more work is needed (as the current generate command expects the external `.proto` files under a vendor folder), but this should at least fix the root-cause: different protobuf generators generating different output.

brocaar, 2019-02-28:
With the migration from dep to Go modules + the above commit, this issues should be fixed.

commit 361e2b24 (2019-02-28): Fix generate API with go modules. Update generated code.

Closes #293.
closed 2019-02-28 by brocaar