brocaar/chirpstack-application-server #295: InfluxDB - panic: runtime error / signal SIGSEGV: segmentation violation

jceloria, 2019-02-14:
### Is this a bug or a feature request?
bug

### What did you expect?
InfluxDB integration

### What happened?
panic: runtime error: invalid memory address or nil pointer dereference
signal SIGSEGV: segmentation violation

### What version are your using?
docker-ce-18.09.2-3.el7.x86_64
lora-app-server: 2.5.0
lora-gateway-bridge: 39c91ac
lora-geo-server: cc6bb5c
loraserver: cbf1ecd
postgres:9.6-alpine
redis:5-alpine
eclipse-mosquitto: 1.5.6

I was using version 2 of the lora* containers before upgrading to latest

I'm using TEKTELIC Kona All-in-One Home Sensors with a Laird RG1xx gateway (Laird Linux gatwick-laird-93.7.3.4) configured with the MQTT forwarder (which frequently disconnects for some reason). I am using a slightly modified JS decoder from https://github.com/SensationalSystems/tektelic-kona-decoder for the payload

### How can your issue be reproduced?
When I trigger an event on the node by placing a magnet near the reed switch sensor, I see the decoded message published to the topic:
```
application/1/device/????????????????/rx {"applicationID":"1","applicationName":"kona","deviceName":"hallway","devEUI":"????????????????","rxInfo":[{"gatewayID":"$$$$$$$$$$$$$$$$","name":"loragw","time":"0001-01-01T00:00:00Z","rssi":-40,"loRaSNR":10.5,"location":{"latitude":0,"longitude":0,"altitude":0}}],"txInfo":{"frequency":904600000,"dr":4},"adr":true,"fCnt":167,"fPort":10,"data":"AQD/CAQACw==","object":{"activity":null,"battery_voltage":null,"bytes":"AQD/CAQACw==","external_input":null,"humidity":null,"reed_count":11,"reed_state":false,"temperature":null}}
```
I then immediately see the panic/SIGSEGV in the logs and the service crashes.

### Could you share your log output?
[log.txt](https://github.com/brocaar/lora-app-server/files/2866516/log.txt)

brocaar, 2019-02-15:
Thanks for reporting and providing logs :-)

I have been able to reproduce this issue and will release a fix asap.

jceloria, 2019-02-15:
Wow, that was fast. Thank you!

commit c87322dd (2019-02-15): Fix panic on nil values in InfluxDB handler.

Closes #295.
closed 2019-02-15 by brocaar