brocaar/chirpstack-application-server #303: Not connected to Websocket api

BobOntario, 2019-03-15:
### Is this a bug or a feature request?
Don't know. only saw error message there and deadloop

### What did you expect?
hope someone can help me out with this issue.  Advice potential cuase root.

### What happened?
When I click "Live Lorawan Frame", i got am error massage say "not connected to websocket"

### What version are your using?
Version I installed few months ago.

### How can your issue be reproduced?
Just click "Live Lorawan Frame"

### Could you share your log output?
root@ecs-de1e-0001:~# journalctl -u lora-app-server -f -n 50
-- Logs begin at Sun 2018-12-23 09:15:01 CST. --
Mar 16 06:04:09 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:04:09+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams received bogus greeting from client: \\\"GET / HTTP/1.1\\\\r\\\\nHost: 11\\\"\""
Mar 16 06:04:37 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:04:37+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams failed to receive the preface from client: EOF\""
Mar 16 06:04:37 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:04:37+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams received bogus greeting from client: \\\"GET / HTTP/1.1\\\\r\\\\nHost: 11\\\"\""
Mar 16 06:04:40 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:04:40+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams received bogus greeting from client: \\\"GET / HTTP/1.1\\\\r\\\\nHost: 11\\\"\""
Mar 16 06:04:41 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:04:41+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams received bogus greeting from client: \\\"GET / HTTP/1.1\\\\r\\\\nHost: 11\\\"\""
Mar 16 06:04:42 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:04:42+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams received bogus greeting from client: \\\"GET / HTTP/1.1\\\\r\\\\nHost: 11\\\"\""
Mar 16 06:04:42 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:04:42+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams received bogus greeting from client: \\\"GET / HTTP/1.1\\\\r\\\\nHost: 11\\\"\""
Mar 16 06:04:42 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:04:42+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams received bogus greeting from client: \\\"GET / HTTP/1.1\\\\r\\\\nHost: 11\\\"\""
Mar 16 06:04:42 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:04:42+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams received bogus greeting from client: \\\"GET / HTTP/1.1\\\\r\\\\nHost: 11\\\"\""
Mar 16 06:04:43 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:04:43+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams received bogus greeting from client: \\\"GET / HTTP/1.1\\\\r\\\\nHost: 11\\\"\""
Mar 16 06:04:43 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:04:43+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams received bogus greeting from client: \\\"GET / HTTP/1.1\\\\r\\\\nHost: 11\\\"\""
Mar 16 06:05:04 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:05:04+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams failed to receive the preface from client: EOF\""
Mar 16 06:05:44 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:05:44+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams received bogus greeting from client: \\\"GET / HTTP/1.1\\\\r\\\\nHost: 11\\\"\""
Mar 16 06:05:55 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:05:55+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams failed to receive the preface from client: EOF\""
Mar 16 06:05:55 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:05:55+08:00" level=warning msg="grpc: Server.Serve failed to create ServerTransport:  connection error: desc = \"transport: http2Server.HandleStreams failed to receive the preface from client: EOF\""
Mar 16 06:09:31 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:09:31+08:00" level=info msg="finished streaming call with code Canceled" error="rpc error: code = Canceled desc = context canceled" grpc.code=Canceled grpc.method=StreamFrameLogs grpc.service=api.GatewayService grpc.start_time="2019-03-16T06:02:13+08:00" grpc.time_ms=438314.66 peer.address="[::1]:57206" span.kind=server system=grpc
Mar 16 06:09:32 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:09:32+08:00" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=List grpc.service=api.ApplicationService grpc.start_time="2019-03-16T06:09:32+08:00" grpc.time_ms=3.5 peer.address="[::1]:57190" span.kind=server system=grpc
Mar 16 06:09:37 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:09:37+08:00" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=Get grpc.service=api.ApplicationService grpc.start_time="2019-03-16T06:09:37+08:00" grpc.time_ms=2.377 peer.address="[::1]:57190" span.kind=server system=grpc
Mar 16 06:09:37 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:09:37+08:00" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=List grpc.service=api.DeviceService grpc.start_time="2019-03-16T06:09:37+08:00" grpc.time_ms=4.68 peer.address="[::1]:57196" span.kind=server system=grpc
Mar 16 06:09:43 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:09:43+08:00" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=Get grpc.service=api.ApplicationService grpc.start_time="2019-03-16T06:09:43+08:00" grpc.time_ms=2.349 peer.address="[::1]:57190" span.kind=server system=grpc
Mar 16 06:09:44 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:09:44+08:00" level=info msg="finished client unary call" grpc.code=OK grpc.method=GetDevice grpc.service=ns.NetworkServerService grpc.time_ms=0.772 span.kind=client system=grpc
Mar 16 06:09:44 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:09:44+08:00" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=Get grpc.service=api.DeviceService grpc.start_time="2019-03-16T06:09:44+08:00" grpc.time_ms=3.694 peer.address="[::1]:57196" span.kind=server system=grpc
Mar 16 06:09:44 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:09:44+08:00" level=info msg="finished client unary call" grpc.code=OK grpc.method=GetDeviceProfile grpc.service=ns.NetworkServerService grpc.time_ms=0.964 span.kind=client system=grpc
Mar 16 06:09:44 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:09:44+08:00" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=Get grpc.service=api.DeviceProfileService grpc.start_time="2019-03-16T06:09:44+08:00" grpc.time_ms=3.525 peer.address="[::1]:57212" span.kind=server system=grpc
Mar 16 06:09:44 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:09:44+08:00" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=List grpc.service=api.DeviceProfileService grpc.start_time="2019-03-16T06:09:44+08:00" grpc.time_ms=4.089 peer.address="[::1]:57212" span.kind=server system=grpc
Mar 16 06:09:45 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:09:45+08:00" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=List grpc.service=api.DeviceProfileService grpc.start_time="2019-03-16T06:09:45+08:00" grpc.time_ms=3.95 peer.address="[::1]:57212" s [... cut ...]
Mar 16 06:21:30 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:21:30+08:00" level=info msg="finished unary call with code NotFound" error="rpc error: code = NotFound desc = object does not exist" grpc.code=NotFound grpc.method=GetLastPing grpc.service=api.GatewayService grpc.start_time="2019-03-16T06:21:30+08:00" grpc.time_ms=2.473 peer.address="[::1]:57206" span.kind=server system=grpc
Mar 16 06:21:31 ecs-de1e-0001 lora-app-server[25717]: time="2019-03-16T06:21:31+08:00" level=info msg="finished client streaming call" grpc.code=OK grpc.method=StreamFrameLogsForGateway grpc.service=ns.NetworkServerService grpc.time_ms=0.023 span.kind=client system=grpc
[image]

brocaar, 2019-03-20:
Which browser are you using?

BobOntario, 2019-03-20:
Chrome

brocaar, 2019-07-02:
I can't reproduce this issue and this seems like a configuration issue. https://forum.loraserver.io/search?q=websocket might contain topics that are helpful.

closed 2019-07-02 by brocaar