brocaar/chirpstack-application-server #304: gRPC Error - Application Create

jcrocker-ecomon, 2019-03-18:
**BUG - gRPC Application API Create (Error)**

Creating a new Application via gRPC fails with Unknown UUID on Client and Server.

**Proto: https://github.com/brocaar/lora-app-server/blob/master/api/service-profile.proto**

_List Service Profiles_ (Get the serviceProfileID from id below)

     { id: '39ebfc1a-ab0a-4f46-9a58-a3d433c7a2ca',
           name: 'EU868-Service-Profile',
           organization_id: '1',
           network_server_id: '1',
           created_at: [Object],
           updated_at: [Object] }

**Proto: https://github.com/brocaar/lora-app-server/blob/master/api/application.proto**

_Create Application_

    { application:
       { name: 'test-1-0',
         description: 'Autogenerated Application - test-1-0',
         organizationID: '1',
         serviceProfileID: '39ebfc1a-ab0a-4f46-9a58-a3d433c7a2ca' } }

**Error (grpc):**

    { Error: uuid: incorrect UUID length:
        at Http2CallStream.call.on (/Volumes/Sync/Sync/CEM/git/CLD-ADMIN-NJS/node_modules/@grpc/grpc-js/build/src/client.js:88:45)
        at Http2CallStream.emit (events.js:187:15)
        at Http2CallStream.endCall (/Volumes/Sync/Sync/CEM/git/CLD-ADMIN-NJS/node_modules/@grpc/grpc-js/build/src/call-stream.js:58:18)
        at handlingTrailers (/Volumes/Sync/Sync/CEM/git/CLD-ADMIN-NJS/node_modules/@grpc/grpc-js/build/src/call-stream.js:147:18)
        at process._tickCallback (internal/process/next_tick.js:68:7)
      code: 2,
      details: 'uuid: incorrect UUID length:',
      metadata: Metadata { internalRepr: Map {} } }

**Log (Lora-App-Server):**

    Mar 14 23:12:00 loraserver lora-app-server[3560]: time="2019-03-14T23:12:00Z" level=error msg="finished unary call with code Unknown" error="rpc error: code = Unknown desc = uuid: incorrect UUID length: " grpc.code=Unknown grpc.method=Create grpc.service=api.ApplicationService grpc.start_time="2019-03-14T23:12:00Z" grpc.time_ms=4.757 peer.address="192.168.0.5:54390" span.kind=server system=grpc
    Mar 14 23:17:48 loraserver lora-app-server[3560]: time="2019-03-14T23:17:48Z" level=error msg="finished unary call with code Unknown" error="rpc error: code = Unknown desc = uuid: incorrect UUID length: " grpc.code=Unknown grpc.method=Create grpc.service=api.ApplicationService grpc.start_time="2019-03-14T23:17:48Z" grpc.time_ms=4.809 peer.address="192.168.0.5:54494" span.kind=server system=grpc
    Mar 14 23:17:48 loraserver lora-app-server[3560]: time="2019-03-14T23:17:48Z" level=error msg="finished unary call with code Unknown" error="rpc error: code = Unknown desc = uuid: incorrect UUID length: " grpc.code=Unknown grpc.method=Create grpc.service=api.ApplicationService grpc.start_time="2019-03-14T23:17:48Z" grpc.time_ms=4.243 peer.address="192.168.0.5:54496" span.kind=server system=grpc
    Mar 14 23:17:48 loraserver lora-app-server[3560]: time="2019-03-14T23:17:48Z" level=error msg="finished unary call with code Unknown" error="rpc error: code = Unknown desc = uuid: incorrect UUID length: " grpc.code=Unknown grpc.method=Create grpc.service=api.ApplicationService grpc.start_time="2019-03-14T23:17:48Z" grpc.time_ms=4.158 peer.address="192.168.0.5:54498" span.kind=server system=grpc
    Mar 14 23:17:48 loraserver lora-app-server[3560]: time="2019-03-14T23:17:48Z" level=error msg="finished unary call with code Unknown" error="rpc error: code = Unknown desc = uuid: incorrect UUID length: " grpc.code=Unknown grpc.method=Create grpc.service=api.ApplicationService grpc.start_time="2019-03-14T23:17:48Z" grpc.time_ms=4.584 peer.address="192.168.0.5:54500" span.kind=server system=grpc

**However, it works when using the API (_/api#!/ApplicationService/Create_)**

brocaar, 2019-03-20:
Please note that you might need to use `organizationId` and `serviceProfileId` as field names in gRPC. That is at least the case in Go. Could you check if this is the case?

jcrocker-ecomon, 2019-03-20:
@brocaar Unfortuantely, this did not work. Sending the below still showed the exact same errors for the lora-app-server and gRPC.

```
{ application:
   { name: 'test-1-0',
     description: 'Autogenerated Application - test-1-0',
     organizationId: '1',
     serviceProfileId: '39ebfc1a-ab0a-4f46-9a58-a3d433c7a2ca' } }
```

brocaar, 2019-08-02:
I'm not sure if there is anything I can do. The web-interface does use the gRPC interface (through a REST to gRPC bridge) and creating applications does work fine.

jcrocker-ecomon, 2019-08-02:
Thanks for the response. I ended up moving away from gRPC to a REST client and everything was working as expected. I'm not a gRPC expert by any means so I assume this was something wrong my side - just weird how all the other gRPC methods I was using were working fine. 

Thanks for all your amazing work on this LoRa project - huge respect !!

closed 2019-08-02 by jcrocker-ecomon