brocaar/chirpstack-application-server #311: HTTP JSON Payload does not include timestamp data

proffalken, 2019-04-24:
### Is this a bug or a feature request?

BUG

### What did you expect?

JSON payload sent by the HTTP integration would include data about when the payload was sent/received

### What happened?

All the fields are there about the data other than the timestamps

### What version are your using?

2.6.0

### How can your issue be reproduced?

Create a new HTTP integration, send the data somewhere you can view the output, note that there are no timestamps in the output

### Could you share your log output?

proffalken, 2019-04-24:
FWIW, this is because I'm using the HTTP integration to send data directly to an Elasticsearch database, and without these fields I can't do any kind of timeseries analysis.

Ideally I'd like to see the following:

1) A field that shows when the packet hit the gateway
2) A field that shows when the packet hit the lora-network-server
3) A field that shows when the HTTP payload was sent to the endpoint

I'll happily settle for (1) or (2) as a starting point though :)

proffalken, 2019-04-24:
Is it related to this:

https://forum.loraserver.io/t/http-integration-post-json-data-has-no-time-property/438

`If your gateway does not have a GPS fix, there will not be any time object in the JSON.`

If so, how do I get this feature if GPS is not available on that gateway?

proffalken, 2019-04-25:
ok, my gateway now has a GPS fix (I purchased a USB GPS Mouse to test this) and I still don't have a timestamp in the JSON from the HTTP integration.

Is this a bug?

brocaar, 2019-05-08:
This might be an issue at the gateway side. Note that the Semtech UDP packet-forwarder only supports the UBLOX GPS module (as it uses some non-NMEA messages to get the GPS Epoch time).

When I have a GPS fix, this is what I see (which is the same message as sent to the HTTP integration):

[image]

closed 2019-09-16 by brocaar