brocaar/chirpstack-application-server #312: Allow more than one type of each Integration provider within an application

proffalken, 2019-04-24:
### Is this a bug or a feature request?

Feature Request

### What did you expect?

To be able to add multiple HTTP Integrations to a single application

### What happened?

You can only have one of each type of integration  per app

### What version are your using?

2.6.0

### How can your issue be reproduced?

Try to add an additional HTTP integration to an application that already has one. 

### Could you share your log output?

N/A

---------

We have a few situations where we want to send the data from a single application via HTTP to multiple remote end-points or via InfluxDB to multiple Influx installations.  At present, as we can only have one of each provider type in an application, we would need to write a "proxy" for each integration, send the data to that, and then setup the proxy to forward the data on to each of the remote end-points.

It would be far more preferable if we could add multiple integrations and let LoRaServer.io manage that instead.

brocaar, 2019-05-01:
This is currently not planned and this would have some impact as it would break the current API.

proffalken, 2019-05-01:
ok, thanks.  The main reason for wanting this was to integrate with https://github.com/FarmOS/FarmOS however we've ended up writing an MQTT library instead https://github.com/mockingbirdconsulting/FarmOSMQTT

If this breaks things, then I'm happy for this issue to be closed and we can revisit it if needed in future?

closed 2019-05-01 by proffalken