brocaar/chirpstack-application-server #313: Unable to reach higher latency network servers after 2.5.0

Will-tm, 2019-04-25:
### Is this a bug or a feature request?
Bug

### What did you expect?
I have deployed an infrastructure on AWS with a single lora-app-server in Europe and multiple lora-server all over the world configured for different LoRaWAN regions

### What happened?
Starting version 2.5.0 all lora-server located in Asia Pacific are unreachable by lora-app-server.
If I go back to 2.4.1 everything works as expected

### What version are your using?
2.5.0+

### How can your issue be reproduced?
lora-app-server deployed on AWS Frankfurt datacenter
lora-server deployed on AWS Seoul datacenter

### Could you share your log output?
Apr 25 06:20:06 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:06Z" level=debug msg="sql query executed" args="[me 161]" duration=3.993231ms query="select count(*) from (\n\tselect 1\n\tfrom \"user\" u\n\tleft join organization_user ou\n\t\ton u.id = ou.user_id\n\tleft join organization o\n\t\ton o.id = ou.organization_id\n\tleft join gateway g\n\t\ton o.id = g.organization_id\n\tleft join application a\n\t\ton a.organization_id = o.id\n\tleft join service_profile sp\n\t\ton sp.organization_id = o.id\n\tleft join device_profile dp\n\t\ton dp.organization_id = o.id\n\tleft join network_server ns\n\t\ton ns.id = sp.network_server_id or ns.id = dp.network_server_id\n\tleft join device d\n\t\ton a.id = d.application_id\n\tleft join multicast_group mg\n\t\ton sp.service_profile_id = mg.service_profile_id\n where ((u.username = $1) and (u.is_active = true) and (u.is_admin = true)) or ((u.username = $1) and (u.is_active = true) and (ou.is_admin = true) and (ns.id = $2)) limit 1) count_only"
Apr 25 06:20:06 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:06Z" level=debug msg="sql query executed" args="[161]" duration=1.003638ms query="select * from network_server where id = $1"
Apr 25 06:20:06 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:06Z" level=warning msg="creating insecure network-server client" server="lora-cn1.xxx:xxx"
Apr 25 06:20:06 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:06Z" level=debug msg="parsed scheme: \"\""
Apr 25 06:20:06 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:06Z" level=debug msg="scheme \"\" not registered, fallback to default scheme"
Apr 25 06:20:06 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:06Z" level=debug msg="ccResolverWrapper: sending new addresses to cc: [{lora-cn1.xxx:xxx 0  <nil>}]"
Apr 25 06:20:06 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:06Z" level=debug msg="ClientConn switching balancer to \"pick_first\""
Apr 25 06:20:06 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:06Z" level=debug msg="Subchannel Connectivity change to CONNECTING"
Apr 25 06:20:06 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:06Z" level=debug msg="pickfirstBalancer: HandleSubConnStateChange: 0xc0007ee690, CONNECTING"
Apr 25 06:20:06 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:06Z" level=debug msg="sql query executed" args="[]" duration="727.798µs" query="\n\t\tselect\n\t\t\tg.*\n\t\tfrom gateway g\n\t\tinner join network_server ns\n\t\t\ton ns.id = g.network_server_id\n\t\twhere\n\t\t\tns.gateway_discovery_enabled = true\n\t\t\tand g.ping = true\n\t\t\tand (g.last_ping_sent_at is null or g.last_ping_sent_at <= (now() - (interval '24 hours' / ns.gateway_discovery_interval)))\n\t\torder by last_ping_sent_at\n\t\tlimit 1\n\t\tfor update"
Apr 25 06:20:07 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:07Z" level=debug msg="Subchannel Connectivity change to SHUTDOWN"
Apr 25 06:20:07 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:07Z" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=Get grpc.service=api.NetworkServerService grpc.start_time="2019-04-25T06:20:06Z" grpc.time_ms=505.681 peer.address="127.0.0.1:43246" span.kind=server system=grpc
Apr 25 06:20:07 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:07Z" level=debug msg="sql query executed" args="[]" duration="689.949µs" query="\n\t\tselect\n\t\t\tg.*\n\t\tfrom gateway g\n\t\tinner join network_server ns\n\t\t\ton ns.id = g.network_server_id\n\t\twhere\n\t\t\tns.gateway_discovery_enabled = true\n\t\t\tand g.ping = true\n\t\t\tand (g.last_ping_sent_at is null or g.last_ping_sent_at <= (now() - (interval '24 hours' / ns.gateway_discovery_interval)))\n\t\torder by last_ping_sent_at\n\t\tlimit 1\n\t\tfor update"
Apr 25 06:20:08 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:08Z" level=debug msg="sql query executed" args="[]" duration="686.952µs" query="\n\t\tselect\n\t\t\tg.*\n\t\tfrom gateway g\n\t\tinner join network_server ns\n\t\t\ton ns.id = g.network_server_id\n\t\twhere\n\t\t\tns.gateway_discovery_enabled = true\n\t\t\tand g.ping = true\n\t\t\tand (g.last_ping_sent_at is null or g.last_ping_sent_at <= (now() - (interval '24 hours' / ns.gateway_discovery_interval)))\n\t\torder by last_ping_sent_at\n\t\tlimit 1\n\t\tfor update"
Apr 25 06:20:09 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:09Z" level=debug msg="sql query executed" args="[]" duration="680.367µs" query="\n\t\tselect\n\t\t\tg.*\n\t\tfrom gateway g\n\t\tinner join network_server ns\n\t\t\ton ns.id = g.network_server_id\n\t\twhere\n\t\t\tns.gateway_discovery_enabled = true\n\t\t\tand g.ping = true\n\t\t\tand (g.last_ping_sent_at is null or g.last_ping_sent_at <= (now() - (interval '24 hours' / ns.gateway_discovery_interval)))\n\t\torder by last_ping_sent_at\n\t\tlimit 1\n\t\tfor update"
Apr 25 06:20:10 ip-172-31-24-110 lora-app-server[29343]: time="2019-04-25T06:20:10Z" level=debug msg="sql query executed" args="[]" duration="674.994µs" query="\n\t\tselect\n\t\t\tg.*\n\t\tfrom gateway g\n\t\tinner join network_server ns\n\t\t\ton ns.id = g.network_server_id\n\t\twhere\n\t\t\tns.gateway_discovery_enabled = true\n\t\t\tand g.ping = true\n\t\t\tand (g.last_ping_sent_at is null or g.last_ping_sent_at <= (now() - (interval '24 hours' / ns.gateway_discovery_interval)))\n\t\torder by last_ping_sent_at\n\t\tlimit 1\n\t\tfor update"

brocaar, 2019-05-06:
The above logs only contain info and debug logs. Could you provide the exact error?

Will-tm, 2019-05-06:
I actually didn't manage to get much more, the only thing I could add is that I get a popup `context deadline exceeded (code: 2)` when I try to go on the Network Server page

brocaar, 2019-07-02:
Is this issue still relevant?

closed 2019-08-02 by brocaar