brocaar/chirpstack-application-server #316: Show version number in Web Interface

physec-rex, 2019-05-06:
### Is this a bug or a feature request?
Feature request

### What did you expect?
I think it is a good idea to show the current version of the lora-app-server in the web interface.
When I created a bug report, I looked for the version number in the web interface and was surprised I could not find it.

### What version are your using?
I am not sure. Docker Container with sha256:76a081aec7b3d208eb11ffb8079c7107af75d569bfa675d9f1664060b2093ef4

brocaar, 2019-05-08:
Please note that you can run `lora-app-server version` :-) See also:

`$ lora-app-server --help`

```
LoRa App Server is an open-source application-server, part of the LoRa Server project
        > documentation & support: https://www.loraserver.io/lora-app-server
        > source & copyright information: https://github.com/brocaar/lora-app-server

Usage:
  lora-app-server [flags]
  lora-app-server [command]

Available Commands:
  configfile  Print the LoRa Application Server configuration file
  help        Help about any command
  version     Print the LoRa App Server version

Flags:
  -c, --config string   path to configuration file (optional)
  -h, --help            help for lora-app-server
      --log-level int   debug=5, info=4, error=2, fatal=1, panic=0 (default 4)

Use "lora-app-server [command] --help" for more information about a command.
```

In case of Docker, I recommend to use the version tags, e.g. `loraserver/lora-app-server:x.y.z` to make sure you pull a specific version.

physec-rex, 2019-05-08:
Thanks for your information. I did not notice that.

However, I still think it is a good idea to be able to see the running version in the web interface for users that do not have access to the server lora-app-server is running on.

henricazottes, 2019-05-24:
Or when you are running it from the Gateway OS... where this command doesn't work :/

brocaar, 2019-06-24:
> Or when you are running it from the Gateway OS... where this command doesn't work :/

It does, try `/opt/loraserver/loraserver --help` :-)

brocaar, 2020-04-17:
This is coming soon with the next Concentratord and ChirpStack Gateway Bridge update:

[image]

closed 2025-05-02 by brocaar