brocaar/chirpstack-application-server #319: Add gateway name to the rest of the events similar to Uplink

KarthikSubramanian2, 2019-05-08:
### Is this a bug or a feature request?
feature request 
### What did you expect?
Expecting gateway name to be present in all events similar to Uplink so that i can add gateway name as a tag while generating metrics (I use my own metric observation pipeline)
### What version are your using?
Latest

brocaar, 2019-07-02:
Wouldn't a correlation ID solve this issue? Usually a location or device-status message comes from an uplink. With a correlation ID you could retrieve these fields from the uplink event :-)

KarthikSubramanian2, 2019-07-22:
That is definitely one way to address this issue. But since we have the gateway MAC, i thought it would be easier for us to pass that ON, or the gateway name so that we don't have to do a lookup with the correlation ID.

closed 2025-05-02 by brocaar