brocaar/chirpstack-application-server #321: Feature request: copy JWT Token in user profile

henricazottes, 2019-05-24:
### Is this a bug or a feature request?
Feature request

I want to automate device creation on the Lora Server when programming a node. Given some information such as the app name, my platformio script would generate needed uuids, create the device on the platform with the REST api and inject the information in the node when flashing.

In order to use the API, you need a JWT token. Actually I just inspect the network on firefox and extract it from the bearer in the request. But wouldn't it be useful to have a noob friendly way to get it? For example in the user profile menu on top right, adding a "Copy JWT Token" option wich would copy the current JWT of the user in the clipboard.

Going further, maybe it would be nice to have the ability to generate JWT tokens directly in the app, where we also could manage them. I find it a bit frustrating to have to do it manually, needing an admin-sys checking the secret key etc...

Just sayin' :)

brocaar, 2019-05-29:
I do want to make this easier, but I'm not sure if this would be the right way as the user JWT token will expire after X time. A better idea might be to provide "application tokens", which can be added / revoked through the web-interface and potentially be bound to a single organization or even an application.

What do you think?

henricazottes, 2019-05-29:
It would be perfect! I was suggesting this quickfix but a more advanced token management would be awesome.

You might want to add permission management per token.  Either simple access/modify option for all resources (organization, application, device, etc...) or more complex with the ability to do it per resource.

sagar-patel-sls, 2019-11-14:
@brocaar 
Also, I have checked API user able to access web-interface with API username and password maybe it's a bug
- API user can only access `chirpstack-application-server` using REST API

Thank you

Niondir, 2020-01-03:
For now we actually have to use username/password to get a token. One can create a technical user for this.

I also see much value in having something like application tokens.

brocaar, 2020-01-03:
I started working on API tokens "last year". This is not finished yet as I have to update all the validators and add new tests, but I'm working on this :)

Niondir, 2020-01-03:
Cool sounds great.
Just a thought: Another option would be to just use the same token secret in another application to generate valid tokens. 
At least in closed systems that might be an option.

brocaar, 2020-02-03:
Please see https://github.com/brocaar/chirpstack-application-server/pull/421 for the progress on this.

commit 31754636 (2020-03-17): Implement generating API keys in within web-interface (#421)

This also includes a refactor of some of the storage functions to simplify code, cleanup old code and to cleanup tests.

Closes #321
closed 2020-03-17 by brocaar