brocaar/chirpstack-application-server #323: Cannot Add Org. User via Web Interface

nigelb, 2019-05-29:
### Is this a bug or a feature request?
Bug
### What did you expect?
To be able to create an `Org. User` from the web interface.
### What happened?
Blank page with nothing on it.
### What version are your using?
Docker quick start: brocaar/loraserver-docker@6f576509924e41d31c9b282518014c49c25a57d6 which uses the docker image `loraserver/lora-app-server:3` I also tested the `loraserver/lora-app-server:latest` image, both versions exhibit the issue.
### How can your issue be reproduced?
1. Start Loraserver _et al._:
```bash
$ git checkout https://github.com/brocaar/loraserver-docker.git
$ cd loraserver-docker
$ sudo docker-compose up
```
2. Browse to http://localhost:8080 and login as `admin`
3. Select `lorawan` organization
4. Select `Org. Users`
5. Press `Add` button

[image]

N.B. Tested with both Firefox and Chrome, OS: Ubuntu
### Could you share your log output?
[docker-3.log](https://github.com/brocaar/lora-app-server/files/3230632/docker-3.log)
[docker-latest.log](https://github.com/brocaar/lora-app-server/files/3230633/docker-latest.log)
[firefox_console.log](https://github.com/brocaar/lora-app-server/files/3230634/firefox_console.log)
[google_chrome_console.log](https://github.com/brocaar/lora-app-server/files/3230635/google_chrome_console.log)

brocaar, 2019-05-29:
Thanks for reporting, I've just pushed a fix.

closed 2019-05-29 by brocaar

Niondir, 2019-07-19:
Can this somehow be added to a stable docker release, or do I have to go with latest?

It's a rather critical feature ;)

commit bc16f651 (2019-05-29): Remove leftover from 2880a928b74ce5d3049a51c4b6ed3f11676bf927.

Closes #323.