brocaar/chirpstack-application-server #329: Feature request: Including spreading factor in Event types: Uplink

demetz, 2019-06-10:
### Is this a bug or a feature request?
feature request

### What did you expect?
the spreading factor should be included in the decoded message, along with the RSSI and the SNR

### What happened?
the message sent by the gateway-bridge still contains the field spreading factor, however it is dropped by the app-server

### What version are your using?
3.0

brocaar, 2019-06-24:
Please note that you can use the `dr` field to retrieve the spreading factor.

demetz, 2019-06-24:
if somebody else stumbles across this: the field can be found in the mqtt-messages in `txInfo`, the translation between data rates and spreading factor can be found [here](https://blog.dbrgn.ch/2017/6/23/lorawan-data-rates/)

closed 2019-06-24 by demetz