brocaar/chirpstack-application-server #332: Application ignores the encoder and decoder placed in the device profile

bryanstein, 2019-06-27:
### Is this a bug or a feature request?

bug

### What did you expect?

The codec feature was moved from the application configuration to the device profile in 3.0.0

I expected the application to decode and encode data if I put the appropriate code in the codec and decoder fields in the device profile.

### What happened?

My data was not decoded or encoded after filling out the codec and decoder in the device profile

### What version are your using?

3.0.0

### How can your issue be reproduced?

1. Create and application in lora-app-server via the web interface
2. Add a device to the application
3. Then add a device profile to the device with the appropriate codec/decoder

Your data will be sent to the application but the decoded fields won't show under device data.
Once you then try to initiate a downlink with the fields you placed in the your encoder you'll get an error

### Could you share your log output?

level=error msg=“handle data-down payload error: no or invalid codec configured for application” application_id=1 dev_eui=[thedeviceid]

bryanstein, 2019-06-27:
I mentioned a work around for this issue here on the forum:
https://forum.loraserver.io/t/downlink-via-mqtt-using-the-object-field-fails/5030/4

brocaar, 2019-09-27:
I'm closing this issue. I believe this has already been addressed but the commit was not referencing this issue. Please let me know if you're still experiencing issues with the latest release.

closed 2019-09-27 by brocaar