brocaar/chirpstack-application-server #334: [Bug]: Addition of new Org. User page is not working in LoRa-App-Server.

vikasgulia119, 2019-07-03:
### Bug

I’m trying to setup LoRa server and LoRa App server. Although LoRa Server & LoRa App server is working, I’m not able not add new org. users through LoRa App server web based User Interface. whenever I try to do this operation it leads to a blank page.
i’ve to return back and refresh it to load the previous page. I tried many times but same problem.

brocaar, 2019-07-03:
Please use the provided template when creating an issue. Without this information I can't help you.

```
### Is this a bug or a feature request?

### What did you expect?

### What happened?

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?
```

vikasgulia119, 2019-07-04:
**### Is this a bug or a feature request?**
Bug: Yes

**### What did you expect?**
I setup and installed the LoRa Server and LoRa App Server. After running its instance, the web based UI is working. i want to create two users: 1 - admin user & 1 local user.
after creating the users, when i try to add the local user to org. user it gives blank web page.
[lora-app-server.log](https://github.com/brocaar/lora-app-server/files/3357567/lora-app-server.log)

**### What happened?**
when i try to add a new user in section org. users, it leads to a blank web page.

**### What version are your using?**
LoRa Server; 3.0.2
LoRa App Server: 3.0.0

**### How can your issue be reproduced?**
After installing the LoRa Server and LoRa app server with default setting as mentioned in the guide. Run the instance on localhost with global user credentials and when try to add a new org.users it will show the blank web page.

**### Could you share your log output?**

Jul 03 20:59:04 vm-loraserver systemd[1]: Started LoRa App Server.
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;starting LoRa App Server&quot; docs=&quot;https://www.loraserver.io/&quot; version=3.0.0
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;storage: setting up storage package&quot;
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;storage: setting up Redis pool&quot;
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;storage: connecting to PostgreSQL database&quot;
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;storage: applying PostgreSQL data migrations&quot;
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;storage: PostgreSQL data migrations applied&quot; count=0
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;integration/mqtt: TLS config is empty&quot;
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;integration/mqtt: connecting to mqtt broker&quot; server=&quot;tcp://localhost:1883&quot;
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;api/as: starting application-server api&quot; bind=&quot;0.0.0.0:8001&quot; ca_cert= tls_cert= tls_key=
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;integration/mqtt: connected to mqtt broker&quot;
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;integration/mqtt: subscribing to tx topic&quot; qos=0 topic=application/+/device/+/tx
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;api/external: starting api server&quot; bind=&quot;0.0.0.0:8080&quot; tls-cert= tls-key=
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;api/external: registering rest api handler and documentation endpoint&quot; path=/api
Jul 03 20:59:05 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:05-07:00&quot; level=info msg=&quot;api/js: starting join-server api&quot; bind=&quot;0.0.0.0:8003&quot; ca_cert= tls_cert= tls_key=
Jul 03 20:59:19 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:19-07:00&quot; level=info msg=&quot;finished unary call with code OK&quot; grpc.code=OK grpc.method=Branding grpc.service=api.InternalService grpc.start_time=&quot;2019-07-03T20:59:19-07:00&quot; grpc.time_ms=0.009 peer.address=&quot;127.0.0.1:59884&quot; span.kind=server system=grpc
Jul 03 20:59:19 vm-loraserver lora-app-server[14749]: time=&quot;2019-07-03T20:59:19-07:00&quot; level=info msg=&quot;finished unary call with code OK&quot; grpc.code=OK grpc.method=Profile grpc.service=api.InternalService grpc.start_time=&quot;2019-07-03T20:59:19-07:00&quot; grpc.time_ms=36.189 peer.address=&quot;127.0.0.1:59884&quot; span.kind=server system=grpc

closed 2019-07-11 by vikasgulia119