brocaar/chirpstack-application-server #340: app-server source build in docker on Raspberry pi

OdysLam, 2019-07-24:
### Is this a bug or a feature request?
bug

### What did you expect?
build lora-app-server on Raspberry pi 3 , running alpine arm64

### What happened?
```
Generating combined Swagger JSON
[lora-appserver]  Generating static files
[lora-appserver]  internal/migrations/migrations.go:1: running "go-bindata": exec: "go-bindata": executable file not found in $PATH
[lora-appserver]  
[lora-appserver]  make: *** [Makefile:49: internal/statics] Error 1
[lora-appserver]  
```
### What version are your using?
latest

### How can your issue be reproduced?
Download balena.io, install it on a raspberry pi 3 (arm64 version) and use source dockerfile.

### Could you share your log output?
[lora-app-server-error.txt](https://github.com/brocaar/lora-app-server/files/3426633/lora-app-server-error.txt)
[Dockerfile.txt](https://github.com/brocaar/lora-app-server/files/3426636/Dockerfile.txt)

brocaar, 2019-07-24:
I believe this is more a feature request than a bug as currently building Docker ARM images is not supported :slightly_smiling_face:

OdysLam, 2019-07-24:
So it can’t be monkey patched?

Would it be easy to build a custom app server that just adds(provision) 1 device and outputs any uplink from that device??

I don’t have any experience with grpc and proto, but I do prototype with python 

Thank you so much for your time!

brocaar, 2019-07-24:
> So it can’t be monkey patched?

I don't see it is not possible. I do distribute ARM binaries (see the compile instructions in the docs). I'm just saying that currently I do not provide Docker files to build ARM images, this would require some work.

OdysLam, 2019-07-24:
I would be wiling to contribute, since I am something of a beginner, would you be able to point me towards the right direction?

I don't seem to be able to find those binaries in the docs though.

Cheers! ( Thanks for the swift reply) :)

OdysLam, 2019-07-24:
I have been trying it to make it work by downloading the .deb binary and installing it without building it from source, the resulting container is close to 700mb! in size, nuts.

brocaar, 2019-07-25:
> [lora-appserver]  internal/migrations/migrations.go:1: running "go-bindata": exec: "go-bindata": executable file not found in $PATH

This should be installed here: https://github.com/brocaar/lora-app-server/blob/master/Makefile#L68

It looks like in your ARM build something is failing. I would look into that :-)

OdysLam, 2019-07-25:
I tried installing it stand-alone but it didn't seem to work, tried both $GOPATH and $PATH
```
# WORKDIR $GOPATH
# RUN go get -u -v github.com/jteeuwen/go-bindata
# RUN go install github.com/jteeuwen/go-bindata
```
I think it doesn't support ARM64 architecture, but google findings are scarce and not persistent.

Managed to install it using the binary on a debian image, close to 180mbs. Not great, not terrible. It's eating close to 10mb of ram, so it's good news that it can be easily run on a raspberry pi.

Would you like to PR the dockerfile? Since it's part of a wider project, I will blog about it at some time in the near future, so I will definitely produce a tutorial of sorts (it took me close to 10 hours to make it work)

Cheers!

brocaar, 2019-08-26:
Have you tried to use the arm32v7 golang and alpine containers, e.g.

```
FROM arm32v7/golang:1.12-alpine AS development

ENV PROJECT_PATH=/lora-app-server
ENV PATH=$PATH:$PROJECT_PATH/build
ENV CGO_ENABLED=0
ENV GO_EXTRA_BUILD_ARGS="-a -installsuffix cgo"

RUN apk add --no-cache ca-certificates make git bash protobuf alpine-sdk nodejs nodejs-npm

RUN mkdir -p $PROJECT_PATH
COPY . $PROJECT_PATH
WORKDIR $PROJECT_PATH

RUN make dev-requirements ui-requirements
RUN make

FROM arm32v7/alpine:latest AS production

WORKDIR /root/
RUN apk --no-cache add ca-certificates
COPY --from=development /lora-app-server/build/lora-app-server .
ENTRYPOINT ["./lora-app-server"]
```

OdysLam, 2019-08-26:
My use-case was for aarch64 and the build failed at a certain go dependency `go-bindata`, which I think was due to incompatibility with the architecture.

Here is my working Dockerfile
```
FROM balenalib/raspberrypi3-64-debian:latest-run

RUN apt-get update && apt-get install -y --no-install-recommends curl 
RUN curl https://artifacts.loraserver.io/packages/3.x/deb/pool/main/l/lora-app-server/lora-app-server_3.2.0_linux_arm64.deb -o lora_app.deb -s \ 
&& dpkg -i lora_app.deb && rm -f lora_app.deb \
&& apt-get -y remove --purge curl \
&& apt-get -y autoclean 
RUN apt-get install -y --no-install-recommends openssh-client
# RUN apt-get install -y --no-install-recommends openssh-client

RUN mkdir -p /usr/home
WORKDIR /usr/home

COPY ./configuration/. /etc/lora-app-server/
COPY init_lora-app-server.sh ./
RUN ["chmod", "+x", "init_lora-app-server.sh"]
# CMD ["./init_lora-app-server.sh"]
CMD ["lora-app-server"]
```

alfonzodi, 2019-11-25:
Hi  @OdysLam !  do you finally sucess to compile the source files project on raspberry ?

OdysLam, 2019-11-25:
Hey @alfonzodi, 

I did not manage to compile it from source but rather settled for building a cointainer that would download and install the binary using apt-get. 

Would you like the dockerfile?

OdysLam, 2019-11-25:
Hey @brocaar, 

Would you like to PR a dockerfile for building the project in arch64 by installing the binaries or you think you could PR a dockerfile that builds from source ?

alfonzodi, 2019-11-26:
Hi @OdysLam ,

   Thanks for your answer, well my goal it's to modified the lorawan source code and then compiled the project source code!
    I managed to do that on ubuntu amd64, and now I try to compile this same source code on a raspberry!
   Even if i can also compile it on raspberry, i can not seem to display API WEB and that's maybe due to  the go-bindata package.
- My issue : [https://github.com/brocaar/chirpstack-application-server/issues/384](url)
- Did your docker install all the environnement that i can use ? 
- If you  did not manage to compile it from source, how you generate your binary ?

OdysLam, 2019-11-26:
Hi,

Here is the Dockerfile I used to pull the binary from the directory that is offered by the project:
https://github.com/OdysLam/thesis-eaas/blob/master/source/balena/primary_rpi/lora_appserver/Dockerfile

To use docker on the Raspberry, I used the https://balena.io project. It's truly amazing!

alfonzodi, 2019-11-26:
cool ! thanks i will try it, did you understand why this binary.deb working well, and compiling source code did not ?

OdysLam, 2019-11-26:
For some reason, bindata cannot compile for aarch64. I have no idea how @brocaar compiled it to create the binary, but dockerfile failed every time.

Perhaps we can look into it and propose a PR?

alfonzodi, 2019-11-26:
yes it's would be good idea ! the brocaar propose you to use arm32v7 golang, maybe it's the solution inspite of we work on arm64!

OdysLam, 2019-11-26:
So we have to built it using arm32v7 and it will play in arm64 os?

Might be relevant : https://github.com/docker-library/golang/issues/196

alfonzodi, 2019-11-26:
maybe we can just cross compiling from another arch and use it like binary !

OdysLam, 2019-11-26:
But from that I read in that issue I mentioned, an arm32 compilation won't play on a 64OS hardware. Am I missing something?

Otherwise, yes we can build in 32 env (image a) and then use image B as basis for the container. Multi image builds I think they are called

alfonzodi, 2019-11-26:
you teach me something hahaha ! i must do some search on subject

alfonzodi, 2019-11-26:
well i think i resolve my issue by cross compile from my ubuntu , so i can edited my source code from my pc and then cross compil for arm !

OdysLam, 2019-11-26:
Can you please elaborate on the process you followed ?
Can we do the same in a dockerfile?

Lamtzidis Odysseas,
Website: odyslam.me
Tel:(+30) 6980950995

closed 2025-05-02 by brocaar