brocaar/chirpstack-application-server #343: API application does not change organizationID

geniv, 2019-07-25:
### Is this a bug or a feature request?
bug: yes

### What did you expect?
I expect change organizationID to my defined

### What happened?
problem is with PUT endpoint PUT /api/applications/{application.id}
example if I set an organizationID to something other than the one stored in the record and send it, all the other recordings will change to the one organizationID

### What version are your using?
api version: 1.0.0

### How can your issue be reproduced?
create an app in an organization via POST. Then create a second organization. And then try changing the application (PUT) in one organization to change the binding to another existing one by using the organizationID column

geniv, 2019-07-25:
I wonder why this endpoint is missing when in the application (port 8080) this functionality is and in the API is not ???

brocaar, 2019-07-29:
It is not supported to move applications between organizations. The reason is that the service-profiles and device-profiles are also assigned to an organization and would need to be re-created.

closed 2019-07-29 by brocaar

geniv, 2019-07-29:
@brocaar ok, thanks! but model description is incorrect:

> organizationID (string, optional): ID of the organization to which the application belongs. ,

brocaar, 2019-08-12:
I have updated the api documentation, see the above commit.