brocaar/chirpstack-application-server #344: API missing endpoint GET multicast-groups/{multicast_group_id}/devices

geniv, 2019-07-25:
### Is this a bug or a feature request?
bug: yes
the bigges bug: YES

### What did you expect?
GET /api/multicast-groups/{multicast_group_id}/devices
endpoint for ADD (post) and for REMOVE (delete) without LIST (GET) is useless.
I expect endpoint: /api/multicast-groups/{multicast_group_id}/devices 
If I call this endpoint I get: Method Not Allowed - this is tottaly wrong

### What happened?
I want list devices in multicast group. How?

### What version are your using?
api version: 1.0.0

geniv, 2019-07-25:
I wonder why this endpoint is missing when in the application (port 8080) this functionality is and in the API is not ???

brocaar, 2019-07-29:
Everything that is possible in the web-interface is possible with the API as the web-interface is built on top of the API :wink: 

What you need to use is this endpoint:

[image]

And filter on `multicastGroupID`.

closed 2019-07-29 by brocaar

geniv, 2019-07-29:
@brocaar ok, thanks, I expect endpoint: `/api/multicast-groups/{multicast_group_id}/devices` never mind, I already understand how it is intended