brocaar/chirpstack-application-server #345: Adding GW on a new server

golovlevviktor, 2019-07-30:
### Is this a bug or a feature request?
bug
### What did you expect?
I want to add gateway to the clear server
### What happened?
After I have added Network-server I wanted to add gateway, but field for network-server was empty and there was no any alert message.

Ability to add network-server appears only after I added service profile.

So It will be cool if there will be alert message like when I want to crete device-profile without service-profile =)

### What version are your using?
3
### How can your issue be reproduced?

### Could you share your log output?

mister4yk, 2019-07-31:
First of all you must add service profile.

golovlevviktor, 2019-07-31:
> First of all you must add service profile.

As you can read above I've found it out already, thanks

brocaar, 2019-08-01:
Thanks, I would consider this more as an improvement than bug (it is an inconvenience, but nothing is broken). Anyway it is a good thing to improve :+1:

commit fc04202f (2019-08-02): Check if org. does have service-profiles on gw create.

Closes #345.
closed 2019-08-02 by brocaar