brocaar/chirpstack-application-server #348: CPU workload caused by PostgreSQL

DHintCo, 2019-08-20:
### Is this a bug or a feature request?
Bug

### What happened?
After starting a test with more than 150 nodes, the CPU workload was almost 200% (red arrow, two cores).
[image]

### What version are your using?
LoRa-App-Server: v2.6.1
PostgreSQL v9.6-alpine

### How can your issue be reproduced?
By unchecking a service user as global admin (in the app-server)

The service user is used to allow the application to transmit downlink to each node. In our test setup of 150 nodes, we saw that the CPU workload increases to almost 200% (red arrow). After unchecking the service user as global admin (green arrow), the CPU workload was decreased. This CPU workload was caused by PostgreSQL which uses the following query. 
[image]
This query takes a lot of processing time. 
Is it possible to simplify the query whether the user is allowed to transmit downlink?

Best regards
DHintCo

brocaar, 2019-08-20:
I believe this issue has already been addressed by #300 (and pull-request #302).

brocaar, 2019-09-16:
I'm closing this issue as I think hit has been addressed by the above changes. I've also started making the query more specific for each validator, as it is not always needed to join in all tables. This should also increase the performance.

closed 2019-09-16 by brocaar