brocaar/chirpstack-application-server #349: Wrong API error code

OdysLam, 2019-08-21:
### Is this a bug or a feature request?
bug
### What did you expect?

When adding a duplicate network server, I expect to get statusCode 409 as in every resource

### What happened?

I get statusCode 500 (internal error). This can lead to problems with program logic when designing around the API. It should be consistent as a system.
(amazing work in any case :P )
### What version are your using?
latest
### How can your issue be reproduced?
add duplicate network server
### Could you share your log output?

OdysLam, 2019-08-22:
After testing, it seems that the issue is wider in terms of consistency:
1) Network server:
`error code: 500`
`error message: gRPC error`

2) service/gateway/device profiles:
No duplicate error, can have as many duplicate profiles as you want

3) device/gateway/organization/application
`error code: 409`
`error message: object already exists`

I have not tested for other use-cases but this inconsistency makes APIs prone to error as one have to individually test each api call and can not presume expected functionality.

I would love to propose a PR but I have no golang knowledge whatsoever.

brocaar, 2019-08-26:
Good one, I believe the expected error should have been 409, see: https://github.com/grpc-ecosystem/grpc-gateway/blob/master/runtime/errors.go#L31.

With regards to 2) could you define a "duplicate device-profile"? The ID is generated by LoRa App Server, so every created profile is therefore considered unique.

brocaar, 2019-08-26:
The above commit should solve your issue + probably solves some other issues too. Please let me know if you find any other inconsistencies so that I can fix these too.

closed 2019-08-26 by brocaar

OdysLam, 2019-08-26:
Hey, thank you for the swift response.

Regarding (2), I believe that profiles that are exact the same should not be allowed in the system, much like any other resource. Although I could miss some use-case that you are aware of.

Moreover, I think it would be helpful to initiate docker support for Raspberry (aarch64, armv7). I have a Dockerfile for aarch64, it's not much since I don't build it from source, but it's something. 

Should I PR?

commit cc229dad (2019-08-26): Propagate gRPC errors so that the correct HTTP status is returned.

Closes #349.