brocaar/chirpstack-application-server #359: UI Error message `object does not exist` when last seen is empty

KarthikSubramanian2, 2019-09-13:
Though it doesn't impact any functionality, there is an error message that pops up when the last seen is empty on devices. 

```
object does not exist(code:5)
```

lora-app-server version : 3.2.1

KarthikSubramanian2, 2019-09-13:
Analyzing the code, this is what it seems to be. 

While displaying the `downlink queue` in the `device details` page, the device activation is being fetched from the database and the error appears when the device has not been activated. 

https://github.com/brocaar/lora-app-server/blob/9614b996010b284fe08bdab019a0bc16d62a767e/internal/api/external/device_queue.go#L167

brocaar, 2019-09-17:
Thanks, I have fixed this in the above commit.

commit 6b86e17f (2019-09-17): Fix not found UI error when device is inactive.

Closes #359.
closed 2019-09-17 by brocaar