brocaar/chirpstack-application-server #362: CID not implemented: McGroupStatusReq

liangyuanpeng, 2019-09-22:
### Is this a bug or a feature request?
bug request

### What did you expect?
rpc success

### What happened?

rpc error

code = Internal desc = handle remote multicast setup command error: CID not implemented: McGroupStatusReq" grpc.code=Internal grpc.method=HandleUplinkData grpc.request.deadline="2019-09-22T08:50:51Z" grpc.service=as.ApplicationServerService grpc.start_time="2019-09-22T08:50:50Z" grpc.time_ms=5.085 peer.address="172.19.0.9:53912" span.kind=server system=grpc

### What version are your using?
3.3.1

### How can your issue be reproduced?
just send sensor msg

### Could you share your log output?

time="2019-09-22T08:50:50Z" level=error msg="finished unary call with code Internal" error="rpc error: code = Internal desc = handle remote multicast setup command error: CID not implemented: McGroupStatusReq" grpc.code=Internal grpc.method=HandleUplinkData grpc.request.deadline="2019-09-22T08:50:51Z" grpc.service=as.ApplicationServerService grpc.start_time="2019-09-22T08:50:50Z" grpc.time_ms=5.085 peer.address="172.19.0.9:53912" span.kind=server system=grpc

brocaar, 2019-09-23:
On which port is the device sending its payloads? Could it be that you are using port 200? In this case it collides with the ports proposed for FUOTA:

https://lora-alliance.org/sites/default/files/2019-04/tr002-fuota_process_summary-v1.0.0.pdf

liangyuanpeng, 2019-09-25:
@brocaar  you are right, thank you for your reply!

closed 2019-09-25 by liangyuanpeng

alexdirty, 2019-10-01:
Hello!

@brocaar, we use a device that is implemented on the specification of LoRaWAN v1.0.1. And this device uses port 201.

Because the specification "LoRaWAN Fragmented Data Block Transport v1.0.0 Specification" is optional, then only those devices that transmit large amounts of data should support it. Therefore, the device we use does not support it.

Specification "LoRaWAN Fragmented Data Block Transport v1.0.0"  is in addition to the specifications of the MAC layer LoRaWAN v1.0.2 and v1.1.

The device we use supports the specification of the MAC level LoRaWAN v1.0.1.

lora-app-server supporting the specification LoRaWAN v1.0.2 and v1.1 must support the operation of devices v1.0.0 and v1.0.1.

Changing the LoRaWAN MAC version in the Device Profile does not produce a result.

Can you fix it or help solve the problem?

`30 16:17:29 loraservers lora-app-server[18237]: time="2019-09-30T16:17:29+03:00" level=error msg="finished unary call with code Internal" error="rpc error: code = Internal desc = handle remote fragmentation session command error: CID not implemented: PackageVersionReq" grpc.code=Internal grpc.method=HandleUplinkData grpc.request.deadline="2019-09-30T16:17:30+03:00" grpc.service=as.ApplicationServerService grpc.start_time="2019-09-30T16:17:29+03:00" grpc.time_ms=4.841 peer.address="127.0.0.1:49572" span.kind=server system=grpc`