brocaar/chirpstack-application-server #369: using dots in env variables is not valid.

krasi-georgiev, 2019-09-30:
the current implementation for all loraserver pieces use dots in env variables which is not valid in bash. This makes it impossible to use lora server with env variables. 
```
export APPLICATION_SERVER.INTEGRATION.MQTT.SERVER=tcp://mosquitto:1883
bash: export: `APPLICATION_SERVER.INTEGRATION.MQTT.SERVER=tcp://mosquitto:1883': not a valid identifier
```

Maybe a good fix would be to use  single and double underscores like:
```
APPLICATION__SERVER_INTEGRATION_MQTT_SERVER
```

brocaar, 2019-10-21:
Yes, I agree this should be fixed, using dots was a mistake as it worked with docker-compose. Double underscores as a separator is fine with me, however for the current major version, this should not break any existing configurations.

Maybe this could be configured through a flag which defaults to `.`, but could be set to `__`?

krasi-georgiev, 2019-10-21:
I will look at the implementation and really hope if can be without an additional flag otherwise will add the flag as you suggest

brocaar, 2019-10-21:
:+1: maybe the it can be registered twice? (e.g. once "FOO.BAR" and once "FOO__BAR"). I'm not sure how `viper` handles this, but it is worth trying.

krasi-georgiev, 2019-10-22:
yep, I am thinking about something along these lines

commit 20c0fd10 (2019-11-04): Deprecate the use of env var names with dots (#374)

Note that this is backwards compatible, but will print a warning in case env. var names with a dot are being used.

Closes #369.
closed 2019-11-04 by brocaar