brocaar/chirpstack-application-server #373: Get Devices REST API returned 504 gateway time-out

w1lldone, 2019-10-18:
### Is this a bug or a feature request?
Maybe it's a misconfiguration.
### What did you expect?
To know what caused this issue.
### What happened?
We have 139 active devices registered in our server. Sometime, the device list is gone from the application details page and the API endpoint returned error 504 (gateway time-out).
[image].
The devices list will come back after restarting redis server. Maybe someone has ecountered the same issue and hopefuly know the solution.

### What version are your using?
v3.3.1

sa-wilson, 2019-10-28:
We get something like this happening as well, although I'm not sure if the gateway is timing out like that or not as I haven't recently checked the dev tools when it's happened. I'll do so next time.

We fix it by restarting lora-app-server. A colleague checked the logs the last time it occured (over the weekend) and nothing made it to the debug log. /api/devices also failed to return (which I assume is the issue with the actual device list).

If restarting redis  also fixes this, then maybe there's an issue with the connection between lora-app-server and redis. I'll check redis next time it happens here.

darkfader, 2020-01-10:
I get 504 reading out the events. Perhaps it takes too long somehow?

Tbronders, 2020-01-13:
Hello there,

We have the same issue, a restart did however fix the problem. The time-out error did only occur with the api/devices. We are currently working on version 3.3.0 and looking to upgrade to the latest chirpstack version!

In the log I get the following errors/warnings:

Jan 13 10:30:46 ********* lora-app-server[1641]: time="2020-01-13T10:30:46+01:00" level=warning msg="influxdb integration: unhandled type!" type_name="[]string"

Jan 13 10:30:27  ********* lora-app-server[1641]: time="2020-01-13T10:30:27+01:00" level=error msg="finished unary call with code Internal" error="rpc error: code = Internal desc = handle received ping error: get ping lookup error: get ping lookup error: redigo: nil returned" grpc.code=Internal grpc.method=HandleProprietaryUplink grpc.service=as.ApplicationServerService grpc.start_time="2020-01-13T10:30:27+01:00" grpc.time_ms=1.17 peer.address="10.249.13.244:40700" span.kind=server system=grpc

Jan 13 10:31:19  ********* lora-app-server[1641]: time="2020-01-13T10:31:19+01:00" level=error msg="handle received ping error: get ping lookup error: get ping lookup error: redigo: nil returned"

Could you enlighten what happens here so we can resolve this problem the next time it occurs.

Thanks in advance!

ramuit44, 2020-02-08:
Hi,
We are facing exactly the same issue with v3.6.1 of Chirpstack.  As described above The device list API call times out.  

We checked the GetDevices method under storage/device.go and manually executed the same SQL query and it returned just fine.  Also, The Postgres connections are all fine, with no locks / no long-running transactions/sessions.  

There no errors in the console except for the below log wrt redigo (redis client lib). Also, another thing to note is once this happens, **even adding a new device (post API)was timing out**.  This happened quite a few times last month and **restarting either the redis or app-server**  fixed the problem temporarily.

Which is strange.  I understand that all the device session keys are stored in redis, but I didn't see any references to redis when GetDevices API is called.  Anyways again after a few days, we encounter the same issues again and we kept restarting to fix the problem.

So obviously feels like some connection issue between Redis and the app server.  Any pointers on this would be much appreciated. 

Note:  we are running the setup on K8S with chip stack dockers images.

ramuit44, 2020-02-18:
Further analysis on that after it happened today. 

When we did CLIENT LIST on the Redis we saw below. Quite a number of connections for the PING. I suspect that was the reason for the redis to become unresponsive. We have added a limit to the active redis connections in the  lora app server yaml. Hopefully, that resolves the issue. 

id=130837 addr=172.30.59.205:55368 fd=151 name= age=514586 idle=28 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=233649 addr=172.30.59.205:37280 fd=227 name= age=5533 idle=1 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=233652 addr=172.30.62.25:47708 fd=228 name= age=5523 idle=4 flags=P db=0 sub=1 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=131340 addr=172.30.59.205:34978 fd=175 name= age=512288 idle=10 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=130504 addr=172.30.59.205:50156 fd=122 name= age=516128 idle=12 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=234400 addr=172.30.59.205:49252 fd=257 name= age=2003 idle=2 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=130217 addr=172.30.59.205:45608 fd=88 name= age=517453 idle=13 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=169198 addr=172.30.59.205:43022 fd=218 name= age=326295 idle=28 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=130270 addr=172.30.59.205:46456 fd=92 name= age=517208 idle=13 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=234657 addr=172.30.59.205:53490 fd=262 name= age=753 idle=1 flags=N db=0 sub=0 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=publish
id=131070 addr=172.30.59.205:59010 fd=168 name= age=513508 idle=3 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=233886 addr=172.30.59.205:41084 fd=236 name= age=4398 idle=14 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=130530 addr=172.30.59.205:50562 fd=126 name= age=516008 idle=11 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=234807 addr=127.0.0.1:35750 fd=263 name= age=6 idle=0 flags=N db=0 sub=0 psub=0 multi=-1 qbuf=26 qbuf-free=32742 obl=0 oll=0 omem=0 events=r cmd=client
id=116571 addr=172.30.59.205:45670 fd=119 name= age=584725 idle=11 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=131149 addr=172.30.59.205:60250 fd=174 name= age=513143 idle=17 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=131402 addr=172.30.59.205:35938 fd=187 name= age=511998 idle=2 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=233891 addr=172.30.59.205:41156 fd=237 name= age=4378 idle=3 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=129298 addr=172.30.59.205:59462 fd=14 name= age=521683 idle=16 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=131584 addr=172.30.59.205:38790 fd=205 name= age=511153 idle=14 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=130465 addr=172.30.59.205:49548 fd=116 name= age=516309 idle=12 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=129754 addr=172.30.59.205:38330 fd=50 name= age=519583 idle=15 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=234370 addr=172.30.59.205:48792 fd=249 name= age=2128 idle=25 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=130478 addr=172.30.59.205:49748 fd=118 name= age=516248 idle=12 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=130626 addr=172.30.59.205:52028 fd=136 name= age=515569 idle=9 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=129845 addr=172.30.59.205:39752 fd=57 name= age=519163 idle=14 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=131649 addr=172.30.59.205:39808 fd=210 name= age=510853 idle=13 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=129624 addr=172.30.59.205:36298 fd=40 name= age=520183 idle=15 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=131636 addr=172.30.59.205:39608 fd=209 name= age=510913 idle=13 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=129611 addr=172.30.59.205:36094 fd=39 name= age=520243 idle=15 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=131179 addr=172.30.59.205:60726 fd=177 name= age=513008 idle=11 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=131288 addr=172.30.59.205:34150 fd=190 name= age=512528 idle=11 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
[... cut ...]
id=131030 addr=172.30.59.205:58400 fd=165 name= age=513693 idle=6 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=234429 addr=172.30.59.205:49666 fd=261 name= age=1878 idle=13 flags=N db=0 sub=0 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=set
id=234133 addr=172.30.59.205:45030 fd=251 name= age=3243 idle=3 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping
id=129350 addr=172.30.59.205:60270 fd=18 name= age=521443 idle=16 flags=P db=0 sub=2 psub=0 multi=-1 qbuf=0 qbuf-free=0 obl=0 oll=0 omem=0 events=r cmd=ping

vfylyk, 2020-02-19:
I looked further into this, based on what @ramuit44 noticed. It seems to be caused by a goroutine + redis connection pool leak. This problem is particularly relevant to my company because we use the troubleshooting tabs quite extensively when dealing with sensor providers as well as when diagnosing network issues.

Here's how I managed to reproduce the excessive number of connections in redis:

**Context:**
A registered device that does not send uplinks for the duration of the test.

**Steps:**
1. Go to the device page in the UI and start a "Device Data" troubleshooting.
2. Close the page after a considerable length of time (e.g. >20 minutes) running the capture.

**Expected result:**
* A websocket connection is estabilished with the app server and stays stable (e.g. wss://lora.outcomex.com.au/api/devices/cb0f000010ffffff/events).
* A call is placed in the system for `StreamEventLogs`.
* As the device never sends an uplink, the UI page stays blank.
* When done, the websocket connection is closed when the page is closed.
* A log message stating that the `StreamEventLogs` gRPC method has finished, similar to `time="2020-02-19T00:39:26Z" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=StreamEventLogs grpc.service=api.DeviceService grpc.start_time="2020-02-19T00:34:23Z" grpc.time_ms=302393.25 peer.address="127.0.0.1:36280" span.kind=server system=grpc`
* When the RPC call finishes, the redis connection is returned to the pool
* The number of client connections in redis is stable at the "max_idle" count after a while.

**What actually happens:**
* The websocket connections are closed in the UI every 60 seconds and a new one is created.
* Each new websocket connection opens a new RPC call to `StreamEventLogs`, which in turn takes a new redis connection from the pool
* The previously closed websocket connections do NOT cause their respective RPC calls to complete and finish, which can be detected by the lack of the expected "finished streaming call" log messages.
* The redis connections are therefore not returned to the pool. The 'GetEventLogForDevice' function in `internal/eventlog/eventlog.go` keeps pinging the unused connections every 30 seconds. When checking "client list" on redis-cli, all the leaked connections are idle for less than 30 seconds, and they accumulate as per the number of non-finished RPCs.
* After too many redis connections are open, the server starts to misbehave and e.g. not show the list of devices.

[image]

Excerpt of GetEventLogForDevice:

```
	// todo: make this a config value?
	ticker := time.NewTicker(30 * time.Second)
	defer ticker.Stop()

loop:
	for {
		select {
		case <-ticker.C:
			if err := psc.Ping(""); err != nil {
				log.WithError(err).Error("subscription ping error")
				break loop
			}
		case <-ctx.Done():
			break loop
		case err := <-done:
			return err
		}
	}
```

If now you add a twist to this scenario, and after a while the device happens to send an uplink, I can see all the hanging RPCs completing all at once, and the list of client connections in redis to be normalised:

```
time="2020-02-19T00:39:26Z" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=StreamEventLogs grpc.service=api.DeviceService grpc.start_time="2020-02-19T00:34:23Z" grpc.time_ms=302393.25 peer.address="127.0.0.1:36280" span.kind=server system=grpc
time="2020-02-19T00:39:26Z" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=StreamEventLogs grpc.service=api.DeviceService grpc.start_time="2020-02-19T00:35:26Z" grpc.time_ms=239846.53 peer.address="127.0.0.1:36280" span.kind=server system=grpc
time="2020-02-19T00:39:26Z" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=StreamEventLogs grpc.service=api.DeviceService grpc.start_time="2020-02-19T00:36:26Z" grpc.time_ms=179821.81 peer.address="127.0.0.1:36280" span.kind=server system=grpc
time="2020-02-19T00:39:26Z" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=StreamEventLogs grpc.service=api.DeviceService grpc.start_time="2020-02-19T00:37:26Z" grpc.time_ms=119713.43 peer.address="127.0.0.1:36280" span.kind=server system=grpc
```

@brocaar Any thoughts on this?

brocaar, 2020-02-25:
Thanks @vfylyk for all your input! :+1: This is really helpful and indicates where the problem is.

> The websocket connections are closed in the UI every 60 seconds and a new one is created.

Do you have a load-balancer or proxy between the client (browser) and server (ChirpStack Application Server). Do you see the same result when there is a direct connection between the client (browser) and server (ChirpStack Application Server)?

What I'm interested in is to see if there might be a kind of "zombie" connection between the load-balancer / proxy and the server, which keeps the go-routine running. Based on your input I will also do some debugging.

vfylyk, 2020-02-25:
> Do you have a load-balancer or proxy between the client (browser) and server (ChirpStack Application Server). Do you see the same result when there is a direct connection between the client (browser) and server (ChirpStack Application Server)?

Good point, I do have load-balancer and proxy. As I run the app server in Kubernetes in Amazon in a private subnet, I use an NGINX reverse proxy as ingress, plus a Network Load Balancer in front of NGINX to make it public.

I'll run some tests tomorrow skipping the proxy and load balancer and will let you know if the same still happens.

vfylyk, 2020-02-26:
Hi @brocaar, I tested the same scenario through `kubectl port-forward`, skipping the AWS load-balancer. 

What happens is the websocket connection does stay stable for as long as I stay on the "Device Data" tab, so no restarts every 60 seconds. However, when I move into another tab and the WS connection is closed, the RPCs still hang (i.e. no "finished streaming call with code OK" log entries). Even if a kill the `kubectl port-forward` process, it makes no difference.

[image]

After the device sends one uplink, then all hanging RPCs close, as happens in the load-balancer scenario:

```
time="2020-02-26T01:55:52Z" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=StreamEventLogs grpc.service=api.DeviceService grpc.start_time="2020-02-26T01:41:04Z" grpc.time_ms=888975.4 peer.address="127.0.0.1:36280" span.kind=server system=grpc
time="2020-02-26T01:55:52Z" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=StreamEventLogs grpc.service=api.DeviceService grpc.start_time="2020-02-26T01:32:16Z" grpc.time_ms=1.4162388e+06 peer.address="127.0.0.1:36280" span.kind=server system=grpc
```

As a comparison, when I use the "Lorawan Frames" tab through the AWS load-balancer, we do see the websocket being restarted every 60 seconds, however whenever the connection is closed in the browser, the "finished call..." log entries appear immediately in the logs, for every one of the closed WS connections:

[image]

```
time="2020-02-26T01:21:41Z" level=warning msg="error reading websocket message: websocket: close 1006 (abnormal closure): unexpected EOF"
time="2020-02-26T01:21:42Z" level=info msg="finished client streaming call" grpc.code=OK grpc.method=StreamFrameLogsForDevice grpc.service=ns.NetworkServerService grpc.time_ms=0.041 span.kind=client system=grpc
time="2020-02-26T01:22:42Z" level=warning msg="error reading websocket message: websocket: close 1006 (abnormal closure): unexpected EOF"
time="2020-02-26T01:22:43Z" level=info msg="finished client streaming call" grpc.code=OK grpc.method=StreamFrameLogsForDevice grpc.service=ns.NetworkServerService grpc.time_ms=0.149 span.kind=client system=grpc
time="2020-02-26T01:23:43Z" level=warning msg="error reading websocket message: websocket: close 1006 (abnormal closure): unexpected EOF"
time="2020-02-26T01:23:44Z" level=info msg="finished client streaming call" grpc.code=OK grpc.method=StreamFrameLogsForDevice grpc.service=ns.NetworkServerService grpc.time_ms=0.044 span.kind=client system=grpc
time="2020-02-26T01:24:44Z" level=warning msg="error reading websocket message: websocket: close 1006 (abnormal closure): unexpected EOF"
time="2020-02-26T01:24:44Z" level=info msg="finished client streaming call" grpc.code=OK grpc.method=StreamFrameLogsForDevice grpc.service=ns.NetworkServerService grpc.time_ms=0.055 span.kind=client system=grpc
```

brocaar, 2020-03-24:
To give you an update, I'm currently refactoring the Redis related code. I'm migrating to godoc.org/github.com/go-redis/redis, which does also support Redis Clustering. It also makes the function subscribing to the Redis Pub/Sub a lot more compact as a lot of magic is handled by the redis library.

However, I don't think this solves the issue. What could help is sending a periodic ping from the server to the WebSocket client (browser). In case there is no response, then the server kills the connection.

It looks like when the server has anything to send to the client (browser), only then if finds out that the client has gone after which the RPC completes. Behavior might be different depending the kind of loadbalancer between the client and server.

brocaar, 2020-03-25:
I have created the following issue: https://github.com/tmc/grpc-websocket-proxy/issues/22.

closed 2020-03-26 by brocaar

brocaar, 2020-03-26:
This should be fixed (see pull-request: https://github.com/tmc/grpc-websocket-proxy/pull/23). As this has not yet been merged, I have updated the `go.mod` file to use a fork of grpc-websocket-proxy which includes this fix, see https://github.com/brocaar/chirpstack-application-server/commit/e3d81ccff5104a30423cb15f3a6a9a9a739196d6.

Sorry that this took a while to track down. One reason why some people were affected and others not is that when you have setup a proxy which periodically kills the websocket connection then this is making it significantly worse. E.g. when it kills the connection after 1 minute of inactivity, then the websocket wrapper might still have 60 internal requests open, each keeping a single Redis connection open.

commit e3d81ccf (2020-03-26): Update grpc-websocket-proxy with request cancel fix.

Closes #373.