brocaar/chirpstack-application-server #375: Master branch UI is broken and doesn't load.

krasi-georgiev, 2019-10-25:
This just shows the swagger dir and doesn't load the login screen.

```
docker run --net=host -it -e NETWORK_SERVER.GATEWAY.BACKEND.MQTT.SERVER="tcp://127.0.0.1:1883" -e POSTGRESQL.DSN="postgres://loraserver_as:loraserver_as@localhost/loraserver_as?sslmode=disable" -e APPLICATION_SERVER.EXTERNAL_API.JWT_SECRET="tcp://127.0.0.1:1883" --rm loraserver/lora-app-server
```
obviously I have all of the other containers running on the host network.

if I do the same for version 3 (loraserver/lora-app-server:3) it all works fine and loads the gui.

brocaar, 2019-10-31:
This might have been a glitch? I can not reproduce this with the latest `loraserver/lora-app-server:latest` image.

krasi-georgiev, 2019-11-10:
ok I have spent a little bit more time on this and I think the problem is only on linux systems

I just tried `docker build .` on linux and it doesn't load the GUI

here are some logs that I think a relevant.

```
Installing UI requirements
npm WARN @typescript-eslint/eslint-plugin@1.6.0 requires a peer of typescript@* but none is installed. You must install peer dependencies yourself.
npm WARN @typescript-eslint/parser@1.6.0 requires a peer of typescript@* but none is installed. You must install peer dependencies yourself.
npm WARN @typescript-eslint/typescript-estree@1.6.0 requires a peer of typescript@* but none is installed. You must install peer dependencies yourself.
npm WARN leaflet.markercluster@1.4.1 requires a peer of leaflet@~1.3.1 but none is installed. You must install peer dependencies yourself.
npm WARN ts-pnp@1.1.2 requires a peer of typescript@* but none is installed. You must install peer dependencies yourself.
npm WARN tsutils@3.14.1 requires a peer of typescript@>=2.8.0 || >= 3.2.0-dev || >= 3.3.0-dev || >= 3.4.0-dev || >= 3.5.0-dev || >= 3.6.0-dev but none is installed. You must install peer dependencies yourself.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/jest-haste-map/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.0.6 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.0.6: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

audited 902566 packages in 12.862s
found 2 vulnerabilities (1 moderate, 1 high)
```

brocaar, 2019-12-17:
I'm closing this issue as I'm not able to reproduce it and have not seen this issue in the "official" releases that can be pulled from https://hub.docker.com/u/chirpstack.

closed 2019-12-17 by brocaar