brocaar/chirpstack-application-server #381: Generated proto files continue to show old repository name

blocknonip, 2019-11-04:
This is not a problem per se, it can just make the dependency management a little confusing when pulling from different named locations (even though they redirect). Presumably this will be fixed in a future regen of files.

Example:

https://github.com/brocaar/chirpstack-application-server/blob/master/api/gateway.pb.go#L9

brocaar, 2019-11-04:
Yes, this will be fixed in a future release and is because I was working on the rename before the loraserver repository was renamed (I could already build the `github.com/brocaar/chirpstack-application-server` package as you can define this in `go.mod`.

My plan is to provide a separate API / SDK repository containing the protobuf files + scripts to generate code for other languages. I think dependency wise this makes it also cleaner :)

blocknonip, 2019-11-04:
Good point. Would it be possible to regenerate them sooner? On master branch is fine. My updates are currently stuck with:

`panic: proto: duplicate enum registered: common.Modulation`

due to the mix of the two repositories and attempting to load my non-common Golang types via the new name.

brocaar, 2019-11-05:
@blocknonip please see:

* https://github.com/brocaar/chirpstack-api and
* https://github.com/brocaar/chirpstack-application-server/tree/protobuf_integrations

The `protobuf_integrations` branch is still WIP, but it already uses the chirpstack-api package.

blocknonip, 2019-11-05:
Thanks, I will give it a spin.

blocknonip, 2019-11-05:
Quick conversion, the new repo works well. I was also able to clean out a lot of indirect dependencies that had been pulled in by the larger application server and network server repos. Closing this.

closed 2019-11-05 by blocknonip

brocaar, 2019-11-05:
:+1:  the indirect dependencies was the main reason to pull all the API definitions out. 

There is still some wrapping up / documentation work to be done, but this also allows you to use Protobuf based integration payloads: https://github.com/brocaar/chirpstack-application-server/blob/protobuf_integrations/cmd/chirpstack-application-server/cmd/configfile.go#L127

The definitions for the AS integration messages are in: https://github.com/brocaar/chirpstack-api/blob/master/protobuf/as/integration/integration.proto.