brocaar/chirpstack-application-server #386: Gateway status icon on Gateways page

shepiloff, 2019-11-19:
### Is this a bug or a feature request?
feature
### What did you expect?
gateway status display with red/green icon near each GW on GW list page,  not only marker color on the GW map page.

brocaar, 2019-12-10:
Maybe an icon might make the table a bit "busy", but a "Last seen" column would be helpful (and possible after the stats refactor that I have done a few versions ago). Then when the last seen timestamp is  e.g. > 5 minutes ago, mark it red.

mmrein, 2020-03-17:
Ok, I added a "Last seen" column to gateway list in same fashion as it is on devices list. I would also  change the "Last seen" info on gateway details page to "Last seen at" value - in the same fashion as in the device details page.

I'm hesitating on the color marking tho mainly because you can quite easily see if the GW has been seen "few seconds ago" or "2 moths ago".

commit 0fa94cba (2020-03-26): Add "Last seen" columnt to gateway list view. (#444)

Closes #386.
closed 2020-03-26 by brocaar