brocaar/chirpstack-application-server #387: API bug POST /api/multicast-groups/{multicast_group_id}/devices

geniv, 2019-11-22:
### Is this a bug or a feature request?
bug: yes

### What did you expect?
I expected exception number 400

### What happened?
I add device XY to multicast group A (OK) and after I add same device XY to multicast group B (OK->but should by FAIL) => that is total fail, one device has one multicast settings.

### What version are your using?
1.0.0

### How can your issue be reproduced?
create device XY, and create two multicast. Add device to first multicast. And then add same device XY to second multicast.

brocaar, 2019-11-22:
Technically a device can be assigned to multiple multicast-groups. The device can identify the multicast-group based on DevAddr / session-key.

This is also implemented in the Remote Multicast Setup specification.

closed 2019-11-22 by brocaar

geniv, 2019-11-22:
OK, thanks