brocaar/chirpstack-application-server #388: API feature GET /api/devices

geniv, 2019-11-22:
### Is this a bug or a feature request?
bug: no
feature: yes

### What did you expect?
I expect response from API request information about multicastGroupID.
Is possible return informatiou about multicastGroupID in response if device already in any group, or no group?

example:

```
{
  "result": [
    {
      "applicationID": "string",
      "description": "string",
      "devEUI": "string",
      "deviceProfileID": "string",
      "deviceProfileName": "string",
      "deviceStatusBattery": 0,
      "deviceStatusBatteryLevel": 0,
      "deviceStatusBatteryLevelUnavailable": true,
      "deviceStatusExternalPowerSource": true,
      "deviceStatusMargin": 0,
      "lastSeenAt": "2019-11-22T14:16:23.862Z",
      "name": "string",
      "multicastGroupID": "string|null"
    }
  ],
  "totalCount": "string"
}
```

sagar-patel-sls, 2020-01-01:
Hii @geniv 
you can use MulticastGroupService API with DevEUI filter

`http://localhost:8080/api/multicast-groups`

geniv, 2020-01-07:
Hi @sagar-patel-sls , 
I know this endpoint, but I want this information in /api/devices in one request, else I must call combine request for mulicast and devices together.

closed 2025-05-02 by brocaar